prusa3d/PrusaSlicer #11212: Sidebar - "Collapse ALL" function

WhiteDragonD, 2023-08-31:
### Description of the bug

I have a .3mf with 40+ swatches for printing, with 4 mods on each.  When opening the slicer, ALL of them are expanded and this makes it hard to find the one I'm looking for or to create a new one.

Solution:  Add ability to Expand or Collapse all stl/obj inside the slicer program on the side bar.

### Project file & How to reproduce

Create a box, add 4 edits/mods/text... now copy and past 50 times.  Save .3mf.  Close.  Open.  They are all expanded.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 10

### Printer model

Prusa i3 mk2.5s

pilot51, 2024-11-21:
I came here for this because I also find it annoying with lots of expandable objects. Currently in my case 70 objects each with 6 sub-items.

However, I found that this is a duplicate of #10199.

WhiteDragonD, 2025-07-07:
Helloooooo?

closed 2026-09-29 by prusaslicer-bot[bot]