prusa3d/PrusaSlicer #11213: Use RGB color options for Preview

WhiteDragonD, 2023-08-31:
### Description of the bug

With HueForge becoming popular, I'm using the Preview tool to look at the different colors on the layers.  When right clicking on the slider to change the filament color, the most common way to get the real color is to google RGB <color> and it gives you the value.  Can you add an in put box to insert a RGB value to put the color on the layer slider?

### Project file & How to reproduce

Open STL, Slice, add manual filament change, right click to open the color dialog.
Add RGB box.  aka when I google Blue RGB, the value is: #0000FF

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 10

### Printer model

Prusa i3 mk2.5s

andersholck, 2023-09-03:
#0000FF is a 8 bit HEX value, not an RGB value.
PrusaSlicer uses the OS system color picker. The windows standard color picker only includes two numerical color system input fields: 8 bit HSL and 8 bit RGB.

If you search the windows store for microsoft powertools, its a free system extension that enables you to pick colours on screen with Win+shift+c and display the colorvalue in HEX, RGB and HSL, so you can pick a color directly from the HueForge screen and get the numerical RGB value to plug into PrusaSlicer in HSL or RGB.

WhiteDragonD, 2023-09-07:
I'm going to write up another enhancement that's more clear into the enhancement request..............................

closed 2026-09-29 by prusaslicer-bot[bot]