prusa3d/PrusaSlicer #11217: travel movment

juckendesAuge, 2023-08-31:
### Description of the bug

please save travel

Solid infill printed from left to right, even though the start and end of the perimeter is on the right?
There you have an extra travel to the left to the other end.

When infilling from layer number (4), the shorter path is chosen.

[image]
[image]
[Pantry Rack Tray.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/12487736/Pantry.Rack.Tray.3mf.zip)

### Project file & How to reproduce

Open the 3mf file, slice it, look at layer number (1) and number (2).
The same behavior can be seen with layer number (3).

Thank you

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.1-rc1+linux-x64-GTK3-202308221615.AppImage

### Operating system

Fedora release 38 (Thirty Eight)

### Printer model

Prusa MK3S+ 0.6mm Nozzle

samarthgca, 2023-09-09:
Do you want to make it so that the printer moves the least??

juckendesAuge, 2023-09-09:
Hello samarthgca,

Wouldn't that make sense? That would reduce printing time and save power and reduce wear.
After all, the extra travel adds up massively in an hour-long print.
I think I read that the travel was reduced in some version of PrusaSlicer, but I don't remember which version it was.
Therefore, I had now wondered.

Tupson444, 2026-09-14:
still relevant