prusa3d/PrusaSlicer #11224: Prusa Slicer 2.6.0 Crashing when slicing Raft

JesuisOriginal, 2023-09-01:
### Description of the bug

While trying to add raft to config, when the Raft Layer is set to 5 or more, and click slice, it always crash at 70%.

### Project file & How to reproduce

Save project
Print Settings -> Skirt and brim -> Brim off, Skirt 5 loops with 3mm distance
Infill -> Density -> 40%, Pattern -> Triangles, Lengh = 0, Combine Infill every 2 layers, Solid Infill every 3
Support and material -> Raft -> Raft Layers = 7
Save Project, Slice -> Crash at 70%
[Dremel Bits.zip](https://github.com/prusa3d/PrusaSlicer/files/12501410/Dremel.Bits.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 11

### Printer model

Creality K1

JesuisOriginal, 2023-09-01:
Ps: 4 Raft Layers also crashed, 3 works fine.

FidelCapo, 2023-09-04:
This issue should be fixed in version 2.6.1 rc2. Can you please test this version to see if everything is ok? Thank you

closed 2026-09-29 by prusaslicer-bot[bot]