prusa3d/PrusaSlicer #11230: "Infernal" (instead of "Internal") Server Error Message

BMacours, 2023-09-02:
### Description of the bug

Modal box states "Infernal ;-) Server Error" instead of "Internal" when PrusSlicer encounters an HTTP 500 Error. 

See image :
[image]

### Project file & How to reproduce

Just put your MK4 into a setting menu (to force the error) and try to upload and print from PusaSlicer.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

Windows 10

### Printer model

MK4

Thynix, 2023-09-04:
This isn't a bug in Prusa Slicer - it's a bug in the MK4's firmware. The MK4 is what's sending the message. [See here.](https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/56a40a0f776bb096468b47f52eb158e0afc84e08/lib/WUI/nhttp/headers.cpp#L43) I've [opened a PR](https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/3244) to fix it.

BMacours, 2023-09-04:
Well spotted! Thank you.

bubnikv, 2023-09-05:
It will be fixed in firmware. Closing.

closed 2023-09-05 by bubnikv

mix579, 2024-04-10:
Error still exists 7 months later, so maybe you want to reopen it?!?!

lukasmatena, 2024-04-11:
The fix is already merged into fw: https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/3250