prusa3d/PrusaSlicer #11236: 2 parameter descriptions in Snap connectors not displayed correctly in cyrillic

AndylgTom, 2023-09-03:
### Description of the bug

Description of these parameters in Snap connectors(Cut tool) not displayed correctly in cyrillic.

#: src/slic3r/GUI/Gizmos/GLGizmoCut.cpp:2341
msgid "Bulge proportion related to radius"
msgstr "Пропорция выпуклости клипсы, связанная с радиусом"

#: src/slic3r/GUI/Gizmos/GLGizmoCut.cpp:2355
msgid "Space proportion related to radius"
msgstr "Пропорция прорези в клипсе, связанная с радиусом"

### Project file & How to reproduce

[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.6.1-rc1

### Operating system

Win 10

### Printer model

never mind

YuSanka, 2023-09-04:
Thanks for your report. (SPE-1858)

lukasmatena, 2023-11-15:
This should be fixed since [2.6.1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.1). Closing.

closed 2023-11-15 by lukasmatena