prusa3d/PrusaSlicer #11238: Extra material when start after color (filament) change

mpgprimo, 2023-09-03:
**Is your feature request related to a problem? Please describe.**
When you do a color change for small details (like small letters), sometimes you have problems because after the filament change you lose a little bit of material and you need a little bit more to start to print

**Describe the solution you'd like**
Add an option to extrude extra material when you start printing after color change (or a pause)

**Describe how it would work**
MAybe this options is 0. But you can set extra mm of filament you whant to extrude at the moment you start printing again over the part. (like 0.5mm)

rtyr, 2023-09-03:
It is already possible. You can add second line to color change gcode script.

mpgprimo, 2023-09-03:
Where i can add this line? (I am sorry, i am new with prusaslicer, i am blind but i cant see where)

rtyr, 2023-09-03:
[image]

mpgprimo, 2023-09-03:
OMG, ok. I did not spect it was on machine parameters.
Thanks!!!!!! Nice help

closed 2023-09-03 by mpgprimo