prusa3d/PrusaSlicer #11239: Crashes when trying to export thumbnail as jpg

cartoon-raccoon, 2023-09-03:
### Description of the bug

When I try to slice with the Thumbnail Format set to JPG, PrusaSlicer crashes with the error:

```text
Wrong JPEG library version: library is 80, caller expects 62
```
I've checked with with `ldd` and PrusaSlicer loads in `libjpeg.so.8`, while I assume PrusaSlicer needs `libjpeg.so.62`.

### Project file & How to reproduce

[3dbenchypetg.zip](https://github.com/prusa3d/PrusaSlicer/files/12506618/3dbenchypetg.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-unknown

### Operating system

Arch Linux

### Printer model

Ender 3 V2 Neo with mriscoc firmware

lukasmatena, 2023-09-03:
From the version number I assume that you have built PrusaSlicer yourself. Do you have the same problem with our AppImage?

cartoon-raccoon, 2023-09-03:
No, I'm using the version available on [Arch Linux's repos](https://archlinux.org/packages/extra/x86_64/prusa-slicer/). I'll try the AppImage and get back to you.

On the topic of building from source, that's actually another problem, when I tried to build PrusaSlicer from source (using prusa-slicer-git in the AUR), CMake gets stuck on step 17 of Download and Verify, when downloading `dep_Boost`. I don't have the actual output on hand right now, do let me know if you need it.

cartoon-raccoon, 2023-09-03:
Update: I ran the GTK3.0 AppImage, that one works perfectly.

lukasmatena, 2023-09-04:
I am sorry, but we will not debug the problem if our statically built AppImage works fine. We use static linking to avoid exactly this kind of issues.

svenstaro, 2023-11-23:
Arch Linux PrusaSlicer maintainer here. We use the process recommended by upstream and use the static build as you can see [here in the PKGBUILD](https://gitlab.archlinux.org/archlinux/packaging/packages/prusa-slicer/-/blob/main/PKGBUILD?ref_type=heads).

The `lddtree` output is as such:
```
lddtree /usr/bin/prusa-slicer
/usr/bin/prusa-slicer (interpreter => /lib64/ld-linux-x86-64.so.2)
    libjpeg.so.8 => /usr/lib/libjpeg.so.8
    libGL.so.1 => /usr/lib/libGL.so.1
        libGLdispatch.so.0 => /usr/lib/libGLdispatch.so.0
        libGLX.so.0 => /usr/lib/libGLX.so.0
    libz.so.1 => /usr/lib/libz.so.1
    libgtk-3.so.0 => /usr/lib/libgtk-3.so.0
        libgmodule-2.0.so.0 => /usr/lib/libgmodule-2.0.so.0
        libharfbuzz.so.0 => /usr/lib/libharfbuzz.so.0
            libfreetype.so.6 => /usr/lib/libfreetype.so.6
            libharfbuzz.so.0 => !!! circular loop !!!
                libbz2.so.1.0 => /usr/lib/libbz2.so.1.0
                libpng16.so.16 => /usr/lib/libpng16.so.16
                libbrotlidec.so.1 => /usr/lib/libbrotlidec.so.1
                    libbrotlicommon.so.1 => /usr/lib/libbrotlicommon.so.1
            libgraphite2.so.3 => /usr/lib/libgraphite2.so.3
        libpangoft2-1.0.so.0 => /usr/lib/libpangoft2-1.0.so.0
        libfribidi.so.0 => /usr/lib/libfribidi.so.0
        libcairo-gobject.so.2 => /usr/lib/libcairo-gobject.so.2
        libatk-1.0.so.0 => /usr/lib/libatk-1.0.so.0
        libepoxy.so.0 => /usr/lib/libepoxy.so.0
        libXi.so.6 => /usr/lib/libXi.so.6
            libXext.so.6 => /usr/lib/libXext.so.6
        libatk-bridge-2.0.so.0 => /usr/lib/libatk-bridge-2.0.so.0
            libatspi.so.0 => /usr/lib/libatspi.so.0
        libcloudproviders.so.0 => /usr/lib/libcloudproviders.so.0
        libtracker-sparql-3.0.so.0 => /usr/lib/libtracker-sparql-3.0.so.0
            libjson-glib-1.0.so.0 => /usr/lib/libjson-glib-1.0.so.0
            libxml2.so.2 => /usr/lib/libxml2.so.2
                liblzma.so.5 => /usr/lib/liblzma.so.5
                libicuuc.so.73 => /usr/lib/libicuuc.so.73
                    libicudata.so.73 => /usr/lib/libicudata.so.73
            libsqlite3.so.0 => /usr/lib/libsqlite3.so.0
        libXfixes.so.3 => /usr/lib/libXfixes.so.3
    libgdk-3.so.0 => /usr/lib/libgdk-3.so.0
        libxkbcommon.so.0 => /usr/lib/libxkbcommon.so.0
        libwayland-client.so.0 => /usr/lib/libwayland-client.so.0
            libffi.so.8 => /usr/lib/libffi.so.8
        libwayland-cursor.so.0 => /usr/lib/libwayland-cursor.so.0
        libwayland-egl.so.1 => /usr/lib/libwayland-egl.so.1
        libXcursor.so.1 => /usr/lib/libXcursor.so.1
            libXrender.so.1 => /usr/lib/libXrender.so.1
        libXdamage.so.1 => /usr/lib/libXdamage.so.1
        libXcomposite.so.1 => /usr/lib/libXcomposite.so.1
        libXrandr.so.2 => /usr/lib/libXrandr.so.2
        libXinerama.so.1 => /usr/lib/libXinerama.so.1
    libpangocairo-1.0.so.0 => /usr/lib/libpangocairo-1.0.so.0
    libpango-1.0.so.0 => /usr/lib/libpango-1.0.so.0
        libthai.so.0 => /usr/lib/libthai.so.0
            libdatrie.so.1 => /usr/lib/libdatrie.so.1
    libcairo.so.2 => /usr/lib/libcairo.so.2
        libxcb.so.1 => /usr/lib/libxcb.so.1
            libXau.so.6 => /usr/lib/libXau.so.6
            libXdmcp.so.6 => /usr/lib/libXdmcp.so.6
        libxcb-render.so.0 => /usr/lib/libxcb-render.so.0
        libxcb-shm.so.0 => /usr/lib/libxcb-shm.so.0
        libpixman-1.so.0 => /usr/lib/libpixman-1.so.0
    libgdk_pixbuf-2.0.so.0 => /usr/lib/libgdk_pixbuf-2.0.so.0
        libtiff.so.6 => /usr/lib/libtiff.so.6
            libzstd.so.1 => /usr/lib/libzstd.so.1
            libjbig.so.2.1 => /usr/lib/libjbig.so.2.1
    libgio-2.0.so.0 => /usr/lib/libgio-2.0.so.0
        libmount.so.1 => /usr/lib/libmount.so.1
            libblkid.so.1 => /usr/lib/libblkid.so.1
    libgobject-2.0.so.0 => /usr/lib/libgobject-2.0.so.0
    libglib-2.0.so.0 => /usr/lib/libglib-2.0.so.0
        libpcre2-8.so.0 => /usr/lib/libpcre2-8.so.0
    libm.so.6 => /usr/lib/libm.so.6
    libc.so.6 => /usr/lib/libc.so.6
    ld-linux-x86-64.so.2 => /lib64/ld-linux-x86-64.so.2
    libX11.so.6 => /usr/lib/libX11.so.6
    libdbus-1.so.3 => /usr/lib/libdbus-1.so.3
        libsystemd.so.0 => /usr/lib/libsystemd.so.0
            libcap.so.2 => /usr/lib/libcap.so.2
            libgcrypt.so.20 => /usr/lib/libgcrypt.so.20
                libgpg-error.so.0 => /usr/lib/libgpg-error.so.0
            liblz4.so.1 => /usr/lib/liblz4.so.1
    libfontconfig.so.1 => /usr/lib/libfontconfig.so.1
        libexpat.so.1 => /usr/lib/libexpat.so.1
    libstdc++.so.6 => /usr/lib/libstdc++.so.6
    libgcc_s.so.1 => /usr/lib/libgcc_s.so.1
```

The problem as - I see it - is exactly that you're not providing a perfect static build. You still require GTK3 (which has its own version of shared libjpeg). I don't think this is an issue with the packaging process from our side. As far as I can see, we're following the upstream instructions exactly. If you remember our last conversation about this, I specifically changed the package to comply with your instructions so that you would support our users.

svenstaro, 2023-11-23:
With v2.7.0, I've found a hack that consists in me deleting the `deps/+JPEG` dir which forces usage of the same shared JPEG linked to GTK3 all the way through. I'll leave the other deps vendored.

tamasmeszaros, 2023-11-28:
Hi @svenstaro 
I've investigated this problem a bit deeper and it looks like the problem arises when a system installation of libjpeg is available at compile time. CMake's find module mixes the headers from the prefix path of the statically built dependencies and the shared library from the system. 

I would recommend you to stick with the workaround you mentioned, but not by deleting the `deps/+JPEG` folder, but setting the CMake var `PrusaSlicer_deps_PLATFORM_PACKAGES=JPEG` when configuring the dependencies. This will have the same effect without any patching. 
You can add more libraries separated by semicolons, like `PrusaSlicer_deps_PLATFORM_PACKAGES=JPEG;TIFF;CURL`

In the meantime I will also try to fix the issue in the PS main project.

svenstaro, 2023-11-28:
Thanks, I changed to `PrusaSlicer_deps_PLATFORM_PACKAGES` in https://gitlab.archlinux.org/archlinux/packaging/packages/prusa-slicer/-/commit/2e0e577af8e28d91bd23651ae73f63e50d545475.

Harvie, 2025-02-10:
> Thanks, I changed to `PrusaSlicer_deps_PLATFORM_PACKAGES` in https://gitlab.archlinux.org/archlinux/packaging/packages/prusa-slicer/-/commit/2e0e577af8e28d91bd23651ae73f63e50d545475.

It seems the problem still persists even in prusa slicer 2.9.0 built with referenced changes...

Harvie, 2025-02-10:
it seems to staticaly complile LOT OF DEPENDENCIES 😢
https://github.com/prusa3d/PrusaSlicer/tree/master/deps
That is rather brutal. For last 15 minutes i've been building curl (which i already have installed in my system as do most other distros i've seen 🤦)
I might consider that a developer choice, that someone wanna bundle everything. But i don't get why does it then complain about mismatch of shared library version? So either link it staticaly and don't rely on shared lib from distro OR compile it against distro version which makes everything compatible. But doing something weird inbetween will break things. I get this might make sense on windows, where is no dependency management tho... 😥

schiele, 2025-04-04:
@Harvie Unfortunately, you can't link everything statically in a complex system like this since there will always be parts that are interacting with complex system interfaces, like the window manager, that are out there in so many variations that you simply cannot handle all the variation in your software.
So you need to cut the line somewhere, but you might break something no matter where you make the cut. The Prusa team made a decision where they think it is best to make that cut, but as we can see this is for sure also not free of problems. Distro packagers tend to cut much closer to the core software to reuse more software that is already included in the distro, and all the security updates and other fixes are handled in a single place. But for sure the producer of the original software is not capable of reproducing all problems in all variations the software might be used out there and thus cannot support everything.
I am more a fan of the latter approach, but I also understand that this way, I am taking the responsibility to deal with issues arising from that approach. When using something that comes in a distro as a user, we need to understand that we need to decide whom to address the issue to. So if we have some technical awareness, as most Arch Linux users probably have to some reasonable degree, we should always try to reproduce the problem with the original vendor delivery first, like the AppImage (or FlatPak these days), as suggested by @lukasmatena above. If the problem persists there as well then it is likely a problem of the original vendor (Prusa in that case). If it is not present there it is the job of the distro packagers to figure out the difference, even if this is not always easy.
Having said all that, I went even more extreme in the direction of using whatever is possible from the distro, such that there should be no version conflicts between two instances of the same library. But for sure this could introduce a compatibility issue between different libraries. My personal experience as a former Distro maintainer in a commercial environment is that this approach usually is more reliable, though.
If you are interested in trying my approach on Arch Linux, then feel free to try building my package from https://gitlab.archlinux.org/archlinux/packaging/packages/prusa-slicer/-/merge_requests/2 and provide feedback there on how it performs with your failure cases. Getting more feedback on which approach works best for most users, the maintainer can likely make the best choice on which approach to take in the future. If you find any issues, I would be happy to hear there since that might also enable me to improve on the package further.

Harvie, 2026-09-02:
still relevant