prusa3d/PrusaSlicer #11244: Flat Surface Ironing

MrBrain295, 2023-09-04:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I often would like to iron all flat surfaces but there is currently no option to do that.
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
An "Iron flat surfaces" option.
**Describe how it would work**
Be as descriptive as your skill level allows you.
You choose that option under ironing and it only irons flat surfaces.
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Not ironing.
**Additional context**
Add any other context or paint drawing about the feature request here.

MrBrain295, 2023-09-05:
> The first setting "All top surfaces" does that.

Yes but I tested it with a be cut and it still tried to iron curved
surfaces.

rtyr, 2023-09-05:
Do you mean this? https://github.com/prusa3d/PrusaSlicer/issues/6928

MrBrain295, 2023-09-05:
Yeah. I didn’t see that when making sure mine wasn’t a duplicate.

rtyr, 2023-09-05:
Ok, I am closing this. Your issue will be linked.

closed 2023-09-05 by rtyr