prusa3d/PrusaSlicer #11245: Modifying INFILL mid print causes Bridging + Top layers printed 1st

dondec123, 2023-09-04:
### Description of the bug

Please check the attached project after Splicing at 47.45.  I'm changing INFILL from 7% to 15% (we always do this on conical parts) to improve the top layer smoothness.

 2.6.1-r2 is now inserting Bridging and Top Solid INFILL Layers before the Infill % change. (???)

Same results if you trigger the Infill % Change using Height Range Modifier.  Exactly the same.

These solid Infill layers simply don't belong there, my parts worked fine without it in the past, and we use this strategy all the time.  

If you feel this methodology is Correct for whatever reason, please provide a check box option to disable it (with default OFF).   Thank you.
[image]
[PS Solid Infill over Infill% Chg.zip](https://github.com/prusa3d/PrusaSlicer/files/12515134/PS.Solid.Infill.over.Infill.Chg.zip)

### Project file & How to reproduce

Load and Slice project.
Slide layer height to 47.45mm
Observe multiple layers of SOLID INFILL (where none is needed).

This did not happen before 2.6.0
[image]
[image]
[PS Solid Infill over Infill% Chg.zip](https://github.com/prusa3d/PrusaSlicer/files/12515170/PS.Solid.Infill.over.Infill.Chg.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1-rc2

### Operating system

Win 11

### Printer model

Formbot Troodon2.0 (voron2.4 clone) w AfterBurner.

rtyr, 2023-09-05:
You have to disable "Interface shells" option.

Can we close the issue?

[image]

dondec123, 2023-09-05:
Thank you for the fast turn around.

YES... Please CLOSE.

ALSO... having spent a career in Quality Assurance, we would call this a "Documentation Bug'... meaning the description didn't help explain the feature very well.

May I recommend creating a Write Up in the online docs about this.  Alternatively, you'll likely get other bug reports for this same issue.

In any case... Thanks again

    Don

[disable]<https://user-images.githubusercontent.com/36745189/265653227-c37c3927-4f64-455e-92c5-b7c57b792f57.png>

closed 2023-09-05 by rtyr