prusa3d/PrusaSlicer #11246: Unwanted solid infill still impossible to remove, since years ... Again and again ...

xTremePower, 2023-09-04:
### Description of the bug

Hello, 

PrusaSlicer are updated really often, and become the perfect slicer .... but since years it's totally impossible to remove these fuk..... unwanted solid infill things  !

No matter if you put 0 insolid infill thresold, if you enable or disable extra perimeter if needed or on overhang ... these plastic eating infills are still there ....

Why in SuperSlicer, the dev managed to fix this ???

Why you don't do anything against this ?

I often print 2/3kg TPU part for my customers .... and with prusa slicer with exactly same settings, perimeters, infill etc, i can have 30% more weight for the same file .... just because of these solid infills ...

I can fix this by adding modifier cubes .... but it's time consuming and really hard to manage well if the part are complicated...

Yes, lower the top and bottom can fix it also ... but if I need 6 bottom and top and 6 wall, I just want 6 of all .. not 4/5 more solid infill layers inside the model

My models print perfectly without these useless filament consuming solid infills, why in 2.6 it's still impossible to disable simply ?
[image]
[image]
[image]

Supermill fixed this since Dec 22, 2021

[ 2.3.57.8](https://github.com/supermerill/SuperSlicer/tree/2.3.57.8) ....

But sadly they SuperSlicer are still on 2.5

Many people complain about this infills on the net, but you never fix this ....

Thanks

### Project file & How to reproduce

just slice and no way to remove solid infill

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.xxx to 2.6.1

### Operating system

Win 11

### Printer model

No matter

xTremePower, 2023-09-04:
This is with SuperSlicer ... a single guy managed to fix this problem perfecty .... 
But in Pursa it's still there 
[image]
[image]

samarthgca, 2023-09-04:
Hi can u pls mark where is unwanted infll is.

xTremePower, 2023-09-05:
> Hi can u pls mark where is unwanted infll is.

Hello, are you serious buddy ?
You don't see the violet solid infill on the first picture ?

Tupson444, 2023-09-05:
I completely agree. The developers missed the importance of having a choice to turn this thing off.

Previously there was "ensure vertical shell thickness" tick box which only switched between two different algorithms (it couldn't be turned off). In recent updates it was changed to a single algorithm which is generally better. But, there is still no way to actually turn it off.

Many issues were opened for this, and the "main" one is #10102.

xTremePower, 2023-09-05:
I know, people open new threads for this same issue, but this problem is there since year .... Supermill fixed this in a week in SuperSlicer, 
Hundreds of people want the same thing : DISABLE this unwanted useless filament consuming time consuming infill if they don't need it ...

I LOVE PrusaSlicer, I use is since years, I can even pay to fix this issue... but now i just concidere to go on an other slicer because this sh't consume my filament ....

On a 3Kg TPU print, this sh'it can many time consume more than 600gr of filament FOR NOTHING, and even worse : introduce useless retractations and paths who can clog the extruder in TPU

Snuff1eupagus, 2023-09-07:
@xTremePower 

This is now and has been 100% possible in Prusa Slicer since before , and after, 2.4.2. Before they removed the button for it, as well as unchecking the button in 2.4.2 to remove this, you also had to adjust your top and bottom layers using modifiers to completely remove the unwanted fill in some situations. None of this matters now as they (THE DEV TEAM) have completely removed the average persons ability to negate this problem. However I want to ensure you that this is still very possible and simple to do in PrusaSlicer 2.6.1 and above, But I can not show you how to do this here because of the way the Dev team treats these workarounds. In the past as I have shown how to accommodate your needs, the Dev team has shown that they would rather delete your ability to do so, then to expound upon it. So I no longer show how to do these things here or on the forums, but rest assured if you have the desire the current version of the software is easily capable. I'M SORRY, truly  that I can not trust to show you how to achieve this here or on the prusa forms, and while I'm sure it's of little comfort to you ,be sure that you can do this in Prusa slicer, your going to have to learn it the hard way. There have been several instances that the Dev team have  proved they would rather remove a workaround in the current software they develop it, as well as several instances that they have proved they would rather ignore a problem that was important to basic slicer development them take the time to develop it. I'd be happy to make more points to this, however I think SS points this out as well as my own PS Dev team experiences. I have made several calls to this point to thee Dev team only to have the workarounds Stamped out instead of left and developed, the software has gone downhill in a big way since 2.4.2

Good luck

vovodroid, 2023-09-29:
> lower the top and bottom can fix it also ... but if I need 6 bottom and top and 6 wall, I just want 6 of all

The question is what exactly you need. You can set 1 bottom and 1 top, and set minimal vertical shell thickness to, let's say, 1mm. Thus you get the same vertical thickness for all walls, including slopes.

xTremePower, 2023-09-29:
> The question is what exactly you need. You can set 1 bottom and 1 top, and set minimal vertical shell thickness to, let's say, 1mm. Thus you get the same vertical thickness for all walls, including [...]

This is absolutly not a fix sorry ....
 what if I need 10 top and bottom layers ? but real top and bottom, like in a box, it not mean i want 10 layer in every single little diagonal shape,
 
 Every slicer in the world can do it well, even SuperSlicer fixed this issue, but sadly don't do updates since a year, and Super Slicer are a fork of PrusaSlicer, so why a single guy can fix this, but no the complete Prusa team in years ?

Snuff1eupagus, 2023-10-01:
@vovodroid 

I'm sorry but I can not do so here, if the Devs Don't know we're good. As soon as we show they they will turn of the possibility.  So , we don't tell them. I'm sorry you can't figure this out, but no Sarcasm was meant, very  literally this can be done and it is clear that xTremePower is one that has no clue.
the problem is showing how here, will end up as an immediate blockage of an impotent capability in PrusaSlicer, we don't tell the  Dev's end of story. No Sarcasm.

vovodroid, 2023-11-03:
@xTremePower , hi,

I managed to completely disable minimal shell thickness mechanism.
Default:
[image]

Disabled:
[image]

[image]

PrusaSlicer use maximal value of solid infill out of bottom/top layers or thickness in mm. My code change allows completely disable this. Does it look good enough for you?

xTremePower, 2023-11-03:
Hello,

 Now set 5 wall + top and bottom to 6 and show me the result

vovodroid, 2023-11-03:
[image]

xTremePower, 2023-11-03:
Strange, now you just show the top layers number, and you don't show your case top inside layers...

But I'm really bored by prusa team, I switched to Orca and found a script to import prusa profiles... At least it works without hack for a simple 2 year old solid infill problem

vovodroid, 2023-11-04:
> you don't show your case top inside layers...

Ok, I modified my solution. I didn't succeed completely to get rid of solid infill and keep real bottom (above bed or bridge), but nevertheless there is improvement.

Default:
[image]

Reduced:
[image]

Default:
[image]

Reduced while keeping bottom:

[image]

[image]

[image]

Actually what I use this feature is adding extra perimeter every second layer, like in SuperSlicer:

[image]

By default Prusa adds solid infill when perimeter is in the air:

[image]

nup002, 2024-12-01:
@smoluch 
Thank you! This did the trick. I can't believe the hoops we have to jump through just to get prusaslicer to respect my wish to have a fixed number of perimeters. Wasted hours on this.

closed 2026-09-29 by prusaslicer-bot[bot]