prusa3d/PrusaSlicer #11247: Paint-on Variable Layer Height

8tm, 2023-09-04:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Extension for Variable Layer Height (used in the same way like Paint-on supports)

**Describe how it would work**
Let's assume we have default layer height 0.3 mm and detailed layers will be printed with selected height (i.e. 0.10 mm).
I would like to paint external layers which should be printed with 0.10 mm (instead of printing selected layer) i.e. for Perimeter, External Perimeter and for Top solid infill.
The rest of model (like solid infill, internal infill, bridges or other) should be printed with default 0.3 mm.

Idea how I would like to select it is identical with Paint-on - I would like to print in better quality (0.10 mm) external parts which I painted but the rest of selected layer (infill etc) should be printed with 0.3 mm.

[image]

**Describe alternatives you've considered**
Selecting external parts, like in case of Measure functionality?
It can be much difficult to do because of details.

8tm, 2023-09-04:
Similar issues:
https://github.com/prusa3d/PrusaSlicer/issues/11099
https://github.com/prusa3d/PrusaSlicer/issues/10691

Mikelodoncz, 2023-09-08:
My idea was same.
https://github.com/prusa3d/PrusaSlicer/issues/10639

closed 2026-09-29 by prusaslicer-bot[bot]