prusa3d/PrusaSlicer #11248: First or both layers temperature set in filament settings not applied to the sliced object

LeirbagGab, 2023-09-04:
### Description of the bug

Hi, 
I am using PrusaSlicer for my Creality k1 and I noticed that when both the first layer and second layer temp are equal in the filament settings the temperature used is the one in my printers start personalized g-code. If the temperature between the first layer and the other layers is different then I only have this issue for the first layer.

Video Proof https://clipchamp.com/watch/E9WXL9OU4MZ

[image]
[image]

About my PrusaSlicer version:
PrusaSlicer
Version:   2.6.1-rc2+win64
Build:     PrusaSlicer-2.6.1-rc2+win64-202309011710

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 11 (version 22621), édition 64-bit
Total RAM size [MB]: 34,280MB
OpenGL installation
GL version:   4.6.0
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 3080/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled

### Project file & How to reproduce

[CanLid2_Soda_54.4mm_prusa-TEMP-BUG.zip](https://github.com/prusa3d/PrusaSlicer/files/12517164/CanLid2_Soda_54.4mm_prusa-TEMP-BUG.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1-rc2+win64

### Operating system

Windows 11 22H2 22621.2215

### Printer model

Creality K1

LeirbagGab, 2023-09-04:
[PrusaSlicer_Temp_Bug_proof 2023.09.05 - 01.38.01-.zip](https://github.com/prusa3d/PrusaSlicer/files/12517210/PrusaSlicer_Temp_Bug_proof.2023.09.05.-.01.38.01-.zip)
Video proof

rtyr, 2023-09-05:
Slicer shows the preview of the final g-code. You have M104 S0 and M140 S0 in the start g-code, so PS shows 0C for the first layer.

You can set your start g-code for example like this:

```
M82 ;absolute extrusion mode
M140 S{first_layer_bed_temperature[0]}
M104 S{first_layer_temperature[0]}
START_PRINT EXTRUDER_TEMP={first_layer_temperature[0]} BED_TEMP={first_layer_bed_temperature[0]}
G92 E0
G92 E0
G1 F2400 E-0.5
```

LeirbagGab, 2023-09-06:
Willl this start gcode set the first layersq temps to the filament's on in my settings ?

rtyr, 2023-09-06:
Yes

closed 2026-09-29 by prusaslicer-bot[bot]