prusa3d/PrusaSlicer #11253: Prusa generates a non-existent overhanging bottom.

Artemonim, 2023-09-06:
### Description of the bug

Prusa generates a non-existent overhanging bottom.

### Project file & How to reproduce

[image]
[image]
[image]

[prusa_2.6.1rc-2_cant_vase_it.zip](https://github.com/prusa3d/PrusaSlicer/files/12537881/prusa_2.6.1rc-2_cant_vase_it.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1-rc2

### Operating system

Windows 11 Pro 22H2 22621.2215

### Printer model

Ender 3

Artemonim, 2023-09-06:
Also check generation of this line, please.

[image]

jmd73, 2023-09-06:
@Artemonim If you disable "Extra perimeters on overhangs", the overhang perimeters are gone.
[image]

I used PrusaSlicer 2.6.1rc-2 on windows 11.

I hope that helps. 

PS. I don't get the extra line with your 3mf.

closed 2026-09-29 by prusaslicer-bot[bot]