prusa3d/PrusaSlicer #11257: automatic alignment and organic supports

TOMASNetr, 2023-09-07:
Hello,
 I have such an observation and an idea. now I was printing a larger number of the same figures that needed supports. And since organic supports are great, I used them.
And now for an observation: for one figure, I drew forced supports and copied the figure onto the printing surface (approx. 12 pieces - so that I don't have to draw supports on each figure separately), I set it to automatically align, then I chose organic supports in the settings and put to slice. After slicing it gave me a first layer collision error due to the first layer of organic supports.
Could it be possible to auto-align multiple parts with larger spacing for organic supports? There is no problem with regular supports (grid). Or should I somehow adjust the settings before the comparison.
I have PrusaSlicer v2.6.0 and i tested it in v2.6.1 too.

Thanks and PrusaSlicer is the best
[image]

neophyl, 2023-09-08:
It’s always done this, there are many closed issues relating to this already, just search closed issues. 
It’s how it’s designed to work. You can either right click on the arrange button and set the spacing to be higher or if you need them spaced together you need to merge the objects together, that way the support will take into account the other Parts of the single Object.

TOMASNetr, 2023-09-08:
Thanks for much, I apologize for my ignorance of advanced alignment options

closed 2023-09-08 by TOMASNetr