prusa3d/PrusaSlicer #11258: Arrange now (2.6.1) ignores the Complete Individual Objects Extruder Clearance when Enabled

Gregregorovich, 2023-09-07:
### Description of the bug

As of 2.6.1 the Extruder Clearance is ignored when the Arrange button is pressed. Previously (<=2.6.0), Arrange took into account the Extruder Clearance.

### Project file & How to reproduce

Have any 2 individual objects on the print bed.
Enable "Complete Individual Objects"
Click Arrange
See overlapping Extruder Clearance circles.

2.6.0 Arrange:
[image]

2.6.1 Arrange:
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1+win64

### Operating system

Windows 10

### Printer model

MK2.5S

Gregregorovich, 2023-09-07:
Appears to be a duplicate of issue #11167 ; is not quite: I managed to arrange them automatically only after right clicking on the Arrange icon and it seeming to force an update to the minimum spacing.

Gregregorovich, 2023-09-07:
I can't seem to reproduce this now that I've right clicked on Arrange, even without saving the project and restarting PrusaSlicer 2.6.1. It seems to have realised that Complete Individual Objects should update the Arrange minimum spacing... Maybe the first time it's right clicked on it's fixed? I didn't know I could right click on it before looking at the previous issue...

yurii-ivanov, 2023-09-07:
I had it also, was simply attempting to place 22 small objects and "arranging" was placing them next to each other as if it would print layer by layer. Had to install 2.6.0 where it worked on first click. MK3S+

Gregregorovich, 2023-09-07:
> I had it also, was simply attempting to place 22 small objects and "arranging" was placing them next to each other as if it would print layer by layer. Had to install 2.6.0 where it worked on first click. MK3S+

Have you tried right clicking on the Arrange button? If that fixes it then at least it's repeatable...

kubispe1, 2023-09-08:
I am not able to reproduce your issue...Win10  2.6.1+win64. I prepared a video to follow your steps. 
If you will face this issue again, you could also make some gif or video + provide 3mf ... 
https://github.com/prusa3d/PrusaSlicer/issues/11167#issuecomment-1695945602

hooman-beein, 2023-09-08:
> Have you tried right clicking on the Arrange button? If that fixes it then at least it's repeatable...

Indeed simply right clicking the arrange(which opens options, but without changing anything) and then left clicking again did spread them out.
here is video: https://youtu.be/NX3tr94_K3Y
STL file(change extension as github doesn't like .stl)
[WindowNetClip_hole-0.6.txt](https://github.com/prusa3d/PrusaSlicer/files/12561908/WindowNetClip_hole-0.6.txt)

Gregregorovich, 2023-09-09:
> 
> https://youtu.be/NX3tr94_K3Y
> 
> After reopening PrusaSlicer after making it work it still works correctly.

This. I mean, I clicked some different buttons, and I don't remember if I fiddled with the algorithm drop-downs (Fast/Accurate/Balanced), but this is essentially what happened and why I can't repeat it without at least trying another computer.

Maybe removing preferences, creating new ones in <=2.6.0, then trying 2.6.1 could repeat this.

apurplecaboose, 2023-09-10:
I had this bug too. I could produce it consistently on every project (even on application relaunch and pc restart) and had to manually move items into position. However right clicking the arrange and changing the settings made the bug disappear permanently even on application relaunch and pc restart. 

Note: Before reading this thread, I have always used the A hotkey to arrange. I didn't even know arrange had a settings menu on right click. Maybe this bug is only reproducible if you use the hotkey?

I updated from 2.6.0 to 2.6.1 using MK3S+ and MK3 on 0.4mm and 0.6 nozzle presets.

tbrock47, 2023-10-06:
I originally submitted #11424 which was closed and pointed to #11258.
After reading through the threads, I see that there is some partial functionality which I do see working in 2.6.1.

I did have another "Feature Request" related to this functionality.

As recommended in [Sequential printing](https://help.prusa3d.com/article/sequential-printing_124589), I would love to see the slicer "Print objects from front to back, from left to right, and from lowest to tallest part to minimize the chance of collision with the extruder and the X-axis."

In my testing with the current implementation, it does attempt to abide by these suggestions. At a very minimum, I believe it should resort the objects to print from front to back, and lowest to tallest. The left-to-right implementation would be a bonus.

lukasmatena, 2023-10-21:
Fixed in [PrusaSlicer 2.7.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-alpha1). Closing.

closed 2023-10-21 by lukasmatena