prusa3d/PrusaSlicer #11259: [FR] Apply random (or shifted) seam for draft shield

vovodroid, 2023-09-07:
When draft shield seams are aligned it can cause to blob, thus increasing chances for draft shield to be knocked by nozzle.

vovodroid, 2023-09-11:
Actually the same should be applied also for support:

[image]

vovodroid, 2023-09-14:
More or less the same happens when printing concentric infill:

[image]

Lukrative525, 2023-12-22:
I came here to actually request the opposite:

When using a draft shield with a 2-into-1 multi-material printer using a purge bucket, similar to the setup described in #10690, it would be helpful if the seams of the draft shield were located as close as possible to the filament change location (which all but necessitates aligning them).

Perhaps if the draft shield had similar options for seam alignment as found in the "Layers and perimeters -> Advanced" section?
The only problem I see with that is there would be no way to place the seam close to the front of the build platform.

I may just need to write a post-processing script to shuffle around the seam positions for the draft shield.

closed 2026-09-29 by prusaslicer-bot[bot]