prusa3d/PrusaSlicer #11264: Prusa 2.6.0 want to make an update with Prusa 2.0.6

Mdz-3D, 2023-09-07:
### Description of the bug

[image]

I installed the new Prusa 2.6.1, when I opened, it offer me to make an update... and when Prusa started with the update I realize that tray to update with PrusaSlicer 2.6.0, so I cancelled it

### Project file & How to reproduce

Run PrusaSlicer 2.6.1
Automatically Prusa open a window offering to install and update
The update window is open, and show the update named PrusaSlicer 2.6.0

[image]

(My english it's not to good, sorry for that)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1 +win64

### Operating system

Windoews 10

### Printer model

do not matter on this case,  but I hace Creality Cr6se

Janys, 2023-09-11:
Hello, we have flushed the cache on our update server. You should see a correct update file (2.6.1) now. Apologies for the inconvenience!

Can you please try the update again and let me know whether it's working now?

Mdz-3D, 2023-09-11:
Dear Janys,

[image]

[image]

[image]

[image]

[image]

Done!! Thank you!

closed 2023-09-11 by Mdz-3D