prusa3d/PrusaSlicer #11267: Program crashing during Slicing

Rayph1n, 2023-09-08:
### Description of the bug

When slicing the object it gets to "Creating infill and crashes". Same project on 2.6.0 worked fine. 

### Project file & How to reproduce

[Pumpkin_03_PRINT.zip](https://github.com/prusa3d/PrusaSlicer/files/12558007/Pumpkin_03_PRINT.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.1+win64

### Operating system

Windows 11

### Printer model

MK3S+

wlcina, 2023-09-08:
Confirm,
Prusa Slicer 2.6.1 is crashing (just closing the app without any message) during Slicing - within Making infill phase (45%) at large count of triangles (got warning above 1M).
the same 3MF file can be sliced with 2.6.0 without  no issues.

lukasmatena, 2023-09-08:
I can confirm too. We will look into it.
SPE-1865

FidelCapo, 2023-09-14:
This issue should be fixed and the fix will be part of the next release.
Closing.

closed 2023-09-14 by FidelCapo