prusa3d/PrusaSlicer #11268: bridge layer height

DomDjoe, 2023-09-08:
### Description of the bug

Prusa is slicing the bridges with a layer height of 0.4mm, the thick bridges setting is disabled and even limiting the extruder layer height to 0.2 the problem continues. The width of the bridge at 0.4 is fixed, it happens at any configured layer height and with a fill of up to 99% this ends up generating
inconsistency in the flow and changing the quality of the outer layer of the part.

[image]

[image]

[image]

### Project file & How to reproduce

[cubo calibração.zip](https://github.com/prusa3d/PrusaSlicer/files/12561096/cubo.calibracao.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

prusa3d_win_2_6_1

### Operating system

Windows 10

### Printer model

Biqu b1

rtyr, 2023-09-08:
Bridging over inner (usually sparse) infill is always printed with "thick bridges", so it is expected behavior. I can understand it can cause some issues with your 99% infill.

DomDjoe, 2023-09-08:
I didn't realize that was expected behavior, but it would be nice to hear a setting to choose that height.

Tupson444, 2026-09-18:
still relevant