prusa3d/PrusaSlicer #11271: The order of nozzle cleaning and bed temp commands causes false readings and bad first layers for MK4

rasssta, 2023-09-09:
### Description of the bug

Using the profile for Prusa MK4IS I have noticed that when printing PETG the first layer is often way too high. This causes bad adhesion to the bed sheet, and the first layer looks bad as well.

I figured out the cause of this being related to the difference in bed temp between PLA and PETG in combination with the order of commands in the g-code. I use 60C for the first layer of PLA and 85C for PETG. These values are important as the nozzle cleaning is performed as soon as the nozzle temp reaches 170C, and when printing PLA this is not a problem as the bed heats up 60C by the time the nozzle reaches 170C.

But with PETG however, the nozzle cleaning "dance" is performed and ready by the time that the bed is around 65-70C, so still 15-20C to heat up. While the bed continues to heat up the nozzle is just sitting there at 170C and waiting, and this causes a slight slight oozing to happen. When the bed reaches 85C and the first layer calibration is performed, there's a tiny amount of filament hanging out of the nozzle and this causes the load cell readings to happen when that oozed filament touches the bed instead of the nozzle itself touching the bed.

The fix for this is quite straight forward, wait for the bed to reach it's final temperature before performing the nozzle cleaning. This eliminates the possibility of oozing happening while waiting for the bed to reach its target temperature after performing the nozzle cleaning.

Original g-code:

```
; probe to clean the nozzle
G1 X{(min(print_bed_max[0], first_layer_print_min[0] + 32) - 32)+32} Y{((first_layer_print_min[1]) - 4)} Z{5} F4800

M302 S160 ; lower cold extrusion limit to 160C

{if filament_type[initial_tool]=="FLEX"}
G1 E-4 F2400 ; retraction
{else}
G1 E-2 F2400 ; retraction
{endif}

M84 E ; turn off E motor

G29 P9 X{(min(print_bed_max[0], first_layer_print_min[0] + 32) - 32)} Y{(max(0, first_layer_print_min[1]) - 4)} W{32} H{4}

{if first_layer_bed_temperature[initial_tool]<=60}M106 S100{endif}

G0 X{(min(print_bed_max[0], first_layer_print_min[0] + 32) - 32)} Y{(max(0, first_layer_print_min[1]) - 4)} Z{40} F10000

M190 S[first_layer_bed_temperature] ; wait for bed temp
```

Modified g-code:
```
M190 S[first_layer_bed_temperature] ; wait for bed temp

; probe to clean the nozzle
G1 X{(min(print_bed_max[0], first_layer_print_min[0] + 32) - 32)+32} Y{((first_layer_print_min[1]) - 4)} Z{5} F4800

M302 S160 ; lower cold extrusion limit to 160C

{if filament_type[initial_tool]=="FLEX"}
G1 E-4 F2400 ; retraction
{else}
G1 E-2 F2400 ; retraction
{endif}

M84 E ; turn off E motor

G29 P9 X{(min(print_bed_max[0], first_layer_print_min[0] + 32) - 32)} Y{(max(0, first_layer_print_min[1]) - 4)} W{32} H{4}

{if first_layer_bed_temperature[initial_tool]<=60}M106 S100{endif}

G0 X{(min(print_bed_max[0], first_layer_print_min[0] + 32) - 32)} Y{(max(0, first_layer_print_min[1]) - 4)} Z{40} F10000
```

The only modification is that the line containing "M190 S[first_layer_bed_temperature] ; wait for bed temp" has been moved from after the nozzle cleaning to before the nozzle cleaning.

I suspect that this issue might cause even more severe issues when printing with materials that require even higher bed temperatures than PETG.

I did a test print with the original g-code, and purposely did not clean the nozzle even though I noted that there were some oozing while it was waiting for the bed to heat up. This is the result:

[image]

[image]

So this "bug" really needs to be addressed.

I'll attach two other pictures. One which was printed without any manual intervention, i.e. filament oozing and no cleaning performed. And one picture where I removed the oozing filament before first layer calibration was performed. I think the pictures talks for themselves ;)

[image]
[image]

### Project file & How to reproduce

G-code supplied in the description.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1+MacOS-arm64

### Operating system

macOS 13.5.2

### Printer model

Prusa MK4IS 5.0.0-alpha4

natbesh, 2023-11-27:
Like this!
The other thing is to possibly heat the nozzle to 240, wait, let it all ooze out & clear  & then drop temp & clean
or maybe do a retract - but hard to know how much ooze there is

ltlowe, 2024-07-16:
I am having 1st layer issues with PETG and a 0.6 nozzle on my MK4.   I posted more detail in a Firmware-Buddy thread. [#3264](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3264)

closed 2026-09-29 by prusaslicer-bot[bot]