prusa3d/PrusaSlicer #11274: FORCED to use draft shield instead of Wipe Tower when using multi-color/material and/or color painted objects. When sliced with draft shield DISABLED, a draft shield will always come up even though Wipe Tower is enabled (or disabled).

joseuzair718, 2023-09-09:
**Is your feature request related to a problem? Please describe.**
For multi-colored or multi-material prints, it seems that draft shields are ALWAYS used, even though it's disabled in the settings. In some situations I'd like to use Wipe Towers.

**Describe the solution you'd like**
A feature to disabled FULLY draft shields for multi-colored or multi-material (maybe for color painted prints as well) so have an option to use a Wipe Tower instead of being FORCED to use draft shields.

**Describe how it would work**
When multi-color/material and/or color painted prints are to be sliced, and if the option or draft shield is disabled, then the draft shield shouldn't show up, because it's DISABLED. Also, allow a wipe tower to work when it's ENABLED.

**Describe alternatives you've considered**
No alternatives, I thought about making a cube and stretching it vertically (did that in Cura), however the draft shield will still show up even though it's disabled in the settings so that won't make a difference.

**Additional context**
Is there a reason why draft shields are forced to be used and no wipe towers when trying to print multi-color/material and/or color painted objects?

[image]

rtyr, 2023-09-09:
As the notification states, the wipe tower is supported only with relative extruder positioning, while you are using absolute positioning. Switch it to relative if you want to use wipe tower (can be set in printer settings).

rtyr, 2023-09-12:
I believe this issue is answered. There is also a duplicate issue on this topic https://github.com/prusa3d/PrusaSlicer/issues/3626.

closed 2023-09-12 by rtyr

joseuzair718, 2023-09-12:
I reviewed #3626 and others mentions in that one as well, and found that if
you place your own cube as a wipe tower it works depending on WHERE you
place the wipe tower, I I’m using that method for now, works great…. Every
layer it prints on my custom wipe tower first and I create 2, one for each
extruder, technically 4, I have 2 stacked on top of each other. One has
extruder 1 at the bottom and extruder 2 at the top, and the other tower has
it vice versa….

I think as long as it’s placed outside of the area of the item you’re
printing and usually towards 0,0 position. Like this:

For me it works perfectly as I used this method in cura as well when using
filaments that ooze a lot…

Hope this helps and can be posted somewhere.

rtyr, 2023-09-12:
I think there is no reason for such workaround, you can simply switch to the relative E mode in the slicer. Printer should handle both relative/absolute posititioning without issues. You may only need to modify the start g-code if you are using some sort of intro line.