prusa3d/PrusaSlicer #11275: Prusa ignores seam restrictions

Artemonim, 2023-09-09:
### Description of the bug

Prusa ignores seam restrictions at this place on this model with this restrictions.

### Project file & How to reproduce

[Bathroom masking hook.zip](https://github.com/prusa3d/PrusaSlicer/files/12566507/Bathroom.masking.hook.zip)
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Windows 11 Pro 22H2

### Printer model

Elegoo Neptune 3 Plus

Artemonim, 2023-09-09:
Also, the slicer has an odd preference for seam placement on this model - in the round position rather than on the corners on the other side.
[image]

Artemonim, 2023-09-09:
LoL, even the blue zone didn't work for the seam
[image] [image]

Artemonim, 2023-09-09:
I reset the variable layer height and it sliced as expected.

Artemonim, 2023-09-09:
Oh, also, in the post above, I was painting the supports, not the seam 🤦‍♂️

ramosglauco, 2023-09-12:
The seams since version 2.4.0 are a big mess.

ramosglauco, 2023-09-12:
Maybe they can fix it in 3 or 4 years. But until then we'll have a bunch of stupid new features.

Jan-Soustruznik, 2024-09-17:
Hello, @Artemonim Could you check the issue, please. Seams placement algorithm was enhancement. Now can be possible that the issue was solved, try to check in latest release please, 2.8.1-rc1(https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.1-rc1). 

Thank you, have a nice day

closed 2026-09-29 by prusaslicer-bot[bot]