prusa3d/PrusaSlicer #11281: Extruder selection ignored when using support material without interface

KyleMaas, 2023-09-10:
### Description of the bug

Trying to print the Sloppy Coiled Rope Bowl model from DaveMakesStuff.  This requires support material.  I don't want to use interface layers, so I left that unchecked.

[image]

However, this disables the extruder selection box in the Multiple Extruders section, which is stuck on 1 when I'm trying to print on extruder 2.

[image]

This shouldn't matter, since I'm not using interface layers anyway, so it should never use extruder 1.  However, in the generated G-Code, it's outputting a `T0` and trying to print using extruder 1 instead, even though the tool paths that are generated are clearly made for extruder 2 (I run the two extruders with different nozzle sizes, so the toolpath spacing is clearly for the 0.6mm nozzle of extruder 2 and not the 0.3mm nozzle of extruder 1).  Also, trying to use a layer height of >0.3mm results in the error `First layer height can't be greater than nozzle diameter`, even though all of the available extruder options are set to extruder 2 which has an 0.6mm nozzle.

If, instead, I enable interface layers, change the extruder on Multiple Extruders to 2, and then disable interface layers, everything behaves as I'd expect.  The layer height error goes away and the generated G-Code uses `T1` instead.

### Project file & How to reproduce

[Sloppy_Coiled_Rope_Bowl_1 (broken).3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/12569003/Sloppy_Coiled_Rope_Bowl_1.broken.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1+linux-x64-GTK3

### Operating system

Gentoo Linux

### Printer model

Heavily-modified MakerFarm i3v with dual extruders

rtyr, 2023-09-11:
Some supports seems to be incorrectly marked as interfaces, therefore it triggers some toolchanges.

Jan-Soustruznik, 2023-09-11:
Hello, @KyleMaas , 

thank you for your report. We try to fix in some next releases. 

Issue created: SPE-1867

Have a nice day

KyleMaas, 2023-09-11:
@Jan-Soustruznik

Awesome.  Thank you!

FidelCapo, 2023-09-13:
This issue should be fixed and the fix will be part of the next release.
Closing.

closed 2023-09-13 by FidelCapo