prusa3d/PrusaSlicer #11286: Prusa auto speed weird behaviour

L8zin, 2023-09-11:
### Description of the bug

When using the auto speed feature, the speed never reaches the maximum allowed set by 'max print speed'. However, like magic, increasing 'max print speed' will increase all speed proportionally, even though they never reached the value in the first place.

### Project file & How to reproduce

Occurs in any file. Simply set all speed to zero and play around with the auto speed setting. For example, setting it to 20mm/s yields speeds as low as 2mm/s in some models. 

[image]
Above image shows values for speed

[image]
Above image shows actual speeds in model

Auto cooling and filament max flowrate both turned off. Machine limits set way high

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1-win64

### Operating system

Windows 10

### Printer model

Modified Ender 3v2 with E3D V6

jamespearson04, 2023-09-11:
Have to agree, infill is printed at the same speed as perimeters, and top layer infill is printed at max speed. Perimeters and top layer should surely be printed at slower speeds than the infill, which should probably be around the maximum.

[image]
[image]
[image]
[image]

pylorak, 2023-10-20:
Possible duplicate of #9044.

Jan-Soustruznik, 2023-10-23:
closed as duplicate issue

closed 2023-10-23 by Jan-Soustruznik