prusa3d/PrusaSlicer #11287: Prusa Slicer (2.6.0 & 2.6.1) can't seem to slice this file

Jinra, 2023-09-11:
### Description of the bug

Prusa Slicer hangs at 45% when slicing the file. 
[image]

### Project file & How to reproduce

The problem seems to be agnostic of settings as using the complete default profile (see screenshot) also yields the same results. Attached is the STL with the issue. Simply slice the file using any settings.

[Body6.zip](https://github.com/prusa3d/PrusaSlicer/files/12579230/Body6.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 and 2.6.1

### Operating system

Windows 11 and 10

### Printer model

Prusa MK4

JohnOCFII, 2023-09-11:
I downloaded and tested slicing this file on my M1/arm Mac.  It also hung at 45%.

Jan-Soustruznik, 2023-09-12:
It seems that the problem is similar with https://github.com/prusa3d/PrusaSlicer/issues/11267

Issue closed as duplicate issue. 

Thank you, have a nice day

closed 2023-09-12 by Jan-Soustruznik