prusa3d/PrusaSlicer #11290: Prusa Slicer is not resizing it's window content when resized while not active.

schorsch3000, 2023-09-12:
### Description of the bug

When i resize the prusaslicer window while it is not active / not on screen, and than select it, the window it self got it's new size, but the window-content does not.

PS: i'm happy to help testing / debuging this issue.

### Project file & How to reproduce

To reproduce that, i run i3-wm.
have 2 containers on screen. one is irrelevant, the other is a tabbed container containing prusa slicer in the background.

close or move the other container, the tabbed container will expand.
then put prusa slicer in the foreground, this will happen:
[image]

You can see: the window itself uses the the full with, as it is suposed to do, but the window-contents doesn't.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.1+linux-x64-GTK3

### Operating system

Debain12

### Printer model

MK4, but that is irrelevant

schorsch3000, 2023-10-20:
same behavoir in PrusaSlicer-2.7.0-alpha1+linux-x64-GTK3-202310191626.AppImage

Jan-Soustruznik, 2023-10-23:
Hello, @schorsch3000 , 

thank you for your report. 

We have reported some similar issue with i3-wm about the window resize: https://github.com/prusa3d/PrusaSlicer/issues/8768

schorsch3000, 2024-02-29:
No change in 2.7.2 :-)

schorsch3000, 2026-09-02:
i hope i am able ti test 3.0 before that bug will be closed :-(