prusa3d/PrusaSlicer #11291: missing single line features

yahbluez, 2023-09-12:
### Description of the bug

The slicer drops the first bridges in this model.

3D preview, be aware of the two first lines:

[image]

After slicing, the two first lines are missing:

[image]

### Project file & How to reproduce

unzip the file, load it into prusa slicer and hit slice.
[PS bug missing lines 230911.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/12588475/PS.bug.missing.lines.230911.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.1+linux-x64-GTK3

### Operating system

manjaro linux

### Printer model

mk3s+

neophyl, 2023-09-12:
Looks like it is because those bridges are 0.19mm high/thick and your layer height is set to 0.2.  So on the ones that are there they are falling enough into on layer or another to slice while on the bottom ones they aren't.  If you change the layer height or even trim the model (sink it slightly) it changes what is picked up and is sliced.
Its basically pot luck as to if they are or not.  
Should all be picked up though as if its not on one layer then it should be on the next.

yahbluez, 2023-09-12:
No that happens also with a line height of 0.2mm and all other lines with the same height are sliced without issues.
The attached file uses 0.2mm and the same issue happens.

[line art 230911.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/12590506/line.art.230911.stl.zip)

neophyl, 2023-09-12:
Change your layer height for slicing to 0.15 or less and it picks them all up with your original file. The bottom bridges are falling in either layer.   
I agree that they should be picked up but that looks like the underlying issue to me. 

So even when you increase the height to 0.2 it’s still split between layer boundaries. I don’t have your original model but I suspect that if you incrementally increase the thickness you will reach a point where it will always pick them up at a 0.2mm layer height, or if you repositioned the bottom ones vertically until they fall onto a layer start that they would also be picked up.

closed 2026-09-29 by prusaslicer-bot[bot]