prusa3d/PrusaSlicer #11293: pronterface removed from 2.6.1

rmeden, 2023-09-13:
### Description of the bug

Just did the upgrade from 2.6.0 to 2.6.1 and pronterface is gone!  There is no reference to the removal in the release notes.

The SD card shuffle is a pain and pointless for small models (most of what I print).   You can say try Octoprint, but have you tried to get a Raspberry Pi?  They've been unavailable for years.

### Project file & How to reproduce

Upgrade from 2.6.0 to 2.6.1.... try and find pronterface

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Windows 10

### Printer model

i3/mk3

kubispe1, 2023-09-13:
We provided just an old version that is not used by the vast majority of our users. 
These days users can select many other solutions so we decided to not pack to installer anymore. 
Sorry for the potential inconvenience.

closed 2023-09-13 by kubispe1

rmeden, 2023-09-13:
What are some of the better alternatives?  I know about Octoprint, but as mentioned Raspberry Pi's are unavailable and it seems wasteful to keep a windows service running all the time.  Are there other simple solutions like pronterface? 

 I've read through the thread that talked about direct serial support in Prusaslicer going away and it didn't talk about much. 

A google search for pronterface alternatives also didn't turn up much.