prusa3d/PrusaSlicer #11297: Own Formula

jakubfrasunek, 2023-09-14:
Possibility to have own formula which result will show after slicing. The field could be shown in info about slicing like filament usage is. 

Possibility to have a field where I could write my own formula with parameters from slicer already, for example I could write in that field something like this:

`(total_estimated_time * 2) + (total_cost * 1.17) + (filament_used_in_grams * 1.5) + (0.48 * 2.37)`

The parameters would be the existing ones in PS, so we could have our own formula where you could put anything you want for your purposes, mainly for me it would be final price calculation for a customer.
The parameters, numbers and formula I used are purely random just for an example.

kubispe1, 2023-09-15:
There could be an available set of placeholder parsers for your calculation. But those values we put just on G-Code...... 
Your Request is specific to filament consumption, right?

jakubfrasunek, 2023-09-15:
No. I really mean to have the ability to make own formula just like I wrote and see the result in slicer after slicing. So you can count with print time, your margin coefficient for example etc. For me it would be something like this:
Cost x 6 + Estimated printing time x 0.25 ... that is my rough final price calculator that customer pays. And I would like to see the result in Sliced Info. So no it is not specific to filament consumption and I don't even know what you mean by that given the fact that I can already set my filament price and see Used Filament ig grams and meters.

jakubfrasunek, 2023-09-15:
For example:
[image]

closed 2026-09-29 by prusaslicer-bot[bot]