prusa3d/PrusaSlicer #11298: locking up when changing color at layer

Deamadar, 2023-09-14:
### Description of the bug

locks up completely when I slice the item and then goto apply a color layer 

### Project file & How to reproduce

[image]

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0

### Operating system

win 11 latest upodate

### Printer model

anycubic max

kubispe1, 2023-09-15:
Sry, but I don't understand this issue. Can you describe your issue, attache 3MF if it is necessary, or make for us some gif or video?

Deamadar, 2023-09-15:
I have resolved it for now I reran the slicer in admin mode

When slicing the item it finishes then I goto the slider to add a color
change.. I get to add but it lockes up....that is best I can do for you
sorry if it dose not help my other issues were addressed in other posts

closed 2026-09-29 by prusaslicer-bot[bot]