prusa3d/PrusaSlicer #11302: Prusa MK4 profile uses minutes to calculate supports for objects diagonally on z-axis

vidarw, 2023-09-14:
### Description of the bug

Prusa MK4 profile uses extreme times (minutes) to calculate supports opposed to 10-15 seconds for the same file on the MK3s+ Profile.
Reproducable by creating something that is longer than the printer and printing diagonally on z-axis.

[slowsupport.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/12611989/slowsupport.3mf.zip)

### Project file & How to reproduce

Slice with supports on.
Transfer settings to MK3 profile and see difference in time.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0+MacOS-arm64,2.6.1+MacOS-arm64

### Operating system

macOS 12.0.1

### Printer model

MK4

kubispe1, 2023-09-15:
I think the reason is not a profile switch but the fact you changed with default settings also Support Style:
MK3 has Grid Supports - which is really quick
MK4 has Snug Supports - by default
I have really, really quickly checked and compared 2.6.1 and 2.5.0, if there is some fundamental calculating time increase, but it does not seem to me. You can also try... We will maintain a support generator, especially for the case of Organic support style during the time.
I am closing this issue for now.

closed 2023-09-15 by kubispe1

vidarw, 2023-09-15:
@kubispe1 yes you are correct that moving to Grid-style fixes the time it takes to calculate yes.

I also discovered yesterday evening that organic supports are by far the best solution for this model.