prusa3d/PrusaSlicer #11305: Stack overflow on malformed step input

deus-ex-silicium, 2023-09-14:
### Description of the bug

Hello, I have found an issue where a stack overflow bug occurs in PrusaSlicer. During the crash an address sanitizer report shows the stack filled with function frame calls to StepShape_OrientedEdge::EdgeEnd() and StepShape_OrientedEdge::EdgeStart() from the OCCTWrapper.so library. I suspect this is a recursive loop gone wild somewhere. I have reproduced the issue by importing the malformed step file on windows and linux using PrusaSlicer 2.6.1. Below you can find the ASAN report and the malformed step file which causes the issue. This could be potentially abused by someone to cause a simple DoS, I would expect PrusaSlicer to give up when trying to import this malformed input and potentially display an error message about the failure.

More importantly, I wanted to ask about the possibility of reporting a security issue in PrusaSlicer with more impact, do you have a preferred way of disclosing the details with the dev team without it becoming public immediately ? I found an email for security issues on the prusa3d.com (security@prusa3d.com) but that seems to be only for the web app. 

Thanks for your time.
```log
STDOUT:
[2023-09-11 16:26:08.968840] [0x00007faf1d690600] [trace]   Initializing StaticPrintConfigs
[32;1m**** ERR StepFile : Incorrect Syntax : Fails Count : 17 ****[0m
[32;1m*** ERR StepReaderData : Unresolved Reference : Fails Count : 32 ***[0m

STDERR:
AddressSanitizer:DEADLYSIGNAL
=================================================================
==12624==ERROR: AddressSanitizer: stack-overflow on address 0x7ffcf5499ff8 (pc 0x7faf1a3e6240 bp 0x00000000000c sp 0x7ffcf549a000 T0)
    #0 0x7faf1a3e6240 in StepShape_OrientedEdge::EdgeEnd() const (PrusaSlicer/build-asan-lto/src/OCCTWrapper.so+0x102d240) (BuildId: f907cc9e25ffae0f)
    #1 0x7faf1a3e6192 in StepShape_OrientedEdge::EdgeStart() const (PrusaSlicer/build-asan-lto/src/OCCTWrapper.so+0x102d192) (BuildId: f907cc9e25ffae0f)
    #2 0x7faf1a3e6242 in StepShape_OrientedEdge::EdgeEnd() const (PrusaSlicer/build-asan-lto/src/OCCTWrapper.so+0x102d242) (BuildId: f907cc9e25ffae0f)
    #3 0x7faf1a3e6192 in StepShape_OrientedEdge::EdgeStart() const (PrusaSlicer/build-asan-lto/src/OCCTWrapper.so+0x102d192) (BuildId: f907cc9e25ffae0f)
    #4 0x7faf1a3e6242 in StepShape_OrientedEdge::EdgeEnd() const (PrusaSlicer/build-asan-lto/src/OCCTWrapper.so+0x102d242) (BuildId: f907cc9e25ffae0f)
    #5 0x7faf1a3e6192 in StepShape_OrientedEdge::EdgeStart() const (PrusaSlicer/build-asan-lto/src/OCCTWrapper.so+0x102d192) (BuildId: f907cc9e25ffae0f)
    #6 0x7faf1a3e6242 in StepShape_OrientedEdge::EdgeEnd() const (PrusaSlicer/build-asan-lto/src/OCCTWrapper.so+0x102d242) (BuildId: f907cc9e25ffae0f)
[...CUT FOR BREVITY...]
SUMMARY: AddressSanitizer: stack-overflow (PrusaSlicer/build-asan-lto/src/OCCTWrapper.so+0x102d240) (BuildId: f907cc9e25ffae0f) in StepShape_OrientedEdge::EdgeEnd() const
==12624==ABORTING
```

### Project file & How to reproduce

1. CTRL+i to open the import file chooser window
2. Select the attached stack-overflow.step file
3. PrusaSlicer crashes

[stack-overflow.zip](https://github.com/prusa3d/PrusaSlicer/files/12613502/stack-overflow.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.1+win64

### Operating system

Windows 10 Pro 22H2 OS Build 19045.3448

### Printer model

Prusa MK3S+

lukasmatena, 2023-09-18:
Thanks, we will look into it. SPE-1866

lukasmatena, 2023-09-19:
This is quite unfortunate. The problem occurs somewhere deep in OCCT (the library we use to load STEP files) and I did not find a way to detect it early enough. The log messages before the crash show that OCCT is actually aware of some problem, but it seems to keep going. I attempted to run the same check separately, but I was not successful as I am completely unfamiliar with the codebase. I did not find any way around it (something like "stop on errors" or similar).

I also attempted to report the bug to OCCT authors, because I believe that's where it should eventually be fixed, but I am unable to use their issue tracker due to some permission issues.
@jbbourgoin I apologize for doing this, but I noticed that you have been active in OCCT issue tracker. Would you be able to verify that this is a problem in OCCT and report it if so? Like I said, I was not able to report it myself. Minimal example to reproduce is below (lacking `#include`s). Loading the [STEP file attached](https://github.com/prusa3d/PrusaSlicer/files/12613502/stack-overflow.zip) recurses infinitely (as shown above) in the `STEPCAFControl_Reader::ReadFile` call. Happens with the latest 7.7.2 (and also in 7.6.2).

```Handle(TDocStd_Document) document;
    Handle(XCAFApp_Application) application = XCAFApp_Application::GetApplication();
    application->NewDocument(path, document);
    STEPCAFControl_Reader reader;
    IFSelect_ReturnStatus stat = reader.ReadFile(path);
```
Thanks.

deus-ex-silicium, 2023-09-19:
Hello, is this (https://github.com/Open-Cascade-SAS/OCCT) where the code for this dependency resides? Am I correct in thinking that there is no simple way to compile this dependency statically into PrusaSlicer? using `-DSLIC3R_STATIC=1` only seems to affect libslic3r.

Also, as for my request for private contact, should I just try emailing security@prusa3d.com or is there a different preferred way?

Thanks.

lukasmatena, 2023-09-20:
> is this (https://github.com/Open-Cascade-SAS/OCCT) where the code for this dependency resides? 

Yes.

> Am I correct in thinking that there is no simple way to compile this dependency statically into PrusaSlicer?

No, it is quite the opposite. Building the dependencies is about a single CMake call, using them is about adding one directory to `CMAKE_PREFIX_PATH`. The process is described in [Building PrusaSlicer](https://github.com/prusa3d/PrusaSlicer/wiki/Building-PrusaSlicer).

> Also, as for my request for private contact, should I just try emailing [security@prusa3d.com](mailto:security@prusa3d.com) or is there a different preferred way?

Sorry about not answering this one. There is no dedicated address for security issues in PrusaSlicer only, but it will get to us if you send it to security@prusa3d.com.

lukasmatena, 2023-10-06:
I again spend some time trying to log in into OpenCASCADE issue tracker to report it, but I was simply not able to. I only have anonymous read-only access.

@vglukhik Would you be able to file this bug report into your system and evaluate whether it indeed is a problem in the library? Sorry for tagging you like this, but creating an account and logging in the issue tracker is a task above my capabilities. Thanks.

@deus-ex-silicium The report you sent reached us and we are currently processing it.

vglukhik, 2023-10-06:
> Отправлено: 6 октября 2023 г. 09:05
> Кому: prusa3d/PrusaSlicer ***@***.***>
> Копия: Vadim GLUKHIKH ***@***.***>; Mention ***@***.***>

Hello!

If you trying to login to Mantis using GutHub account, I will not reach success, sorry. If you want use Mantis, you must create separated account.

What issue you are talking about? I can create bug report instead of you if you will provide me some more information.

Also you can ask @Dmitry ***@***.***>. He will manage you issue and resolve it with pleasure.

---

Best regards,

Vadim Glukhikh

DevOps engineer  | Open Cascade

Mob.: +7 916 263 96 06

Mob.: +351 915 883 256

www.opencascade.com<http://www.opencascade.com/>,

www.capgemini.com<http://www.capgemini.com/>

[cid:b943a724-8cbc-43af-8a65-eb53c0779924]

Please consider the environment and do not print this email unless absolutely necessary. Open Cascade encourages environmental awareness.

CONFIDENTIALITY NOTICE: This message contains information that may be privileged or confidential. It is intended only for the person to whom it is addressed. If you are not the intended recipient, you are not authorized to read, print, retain, copy, disseminate, distribute, or use this message or any part thereof. If you receive this message in error, please notify the sender immediately and delete all copies of this message.

lukasmatena, 2023-11-02:
@vglukhik Thanks. The issue I am talking about is described above in this post: https://github.com/prusa3d/PrusaSlicer/issues/11305#issuecomment-1724974314. As far as I can tell, it is an infinite recursion on a malformed STEP file. Some errors are detected before that, but it seems there is no way to bail out on them.

closed 2026-09-29 by prusaslicer-bot[bot]