prusa3d/PrusaSlicer #11306: UI is not updated immediately when using keyboard shortcut to change extruder

Jahor, 2023-09-15:
### Description of the bug

When selecting extruder using keyboard shortcuts (as implemented by #6216) new extruder is not shown till focus is moved away

### Project file & How to reproduce

1. Set Printer Settings -> General -> Extruders = 2
2. Load object
3. Select object in the list
4. Press 2 on the keyboard to change extruder (nothing happens visually)
5. Move focus to something else (either another object/part or anywhere else)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.1+MacOS-arm64-202309060800

### Operating system

macOS Ventura Version 13.5.2 (Build 22G91)

### Printer model

Prusa MK4

Jan-Soustruznik, 2023-09-15:
Hello, @Jahor , 

thank you for you report.  I dont know if I understood correctly. Do you expect that the model color will be changed imediently after change extruder by the hotkeys? 

BR

Jahor, 2023-09-15:
Correct. At the moment when you press a key nothing happens at all
(visually), so it looks like it did not work.

Jan-Soustruznik, 2023-09-15:
But you have a selected model, so the color of model in scene is green, when unselect model, then model will have color according to extruder color.

Jahor, 2023-09-15:
That's ok. But it should change the extruder color in the object list
immediately.

Jan-Soustruznik, 2023-09-15:
> That's ok. But it should change the extruder color in the object list immediately.

Yes, this is exactly what is happening.

I closed issue as solved.

closed 2023-09-15 by Jan-Soustruznik