prusa3d/PrusaSlicer #11312: Prusa Slicer 2.6.1 crashes when slicing big .stl file

m00nSec, 2023-09-16:
### Description of the bug

When I try to slice a big .stl file using the default Prusa MK4 Input shaping Profile, PrusaSlilcer crashes without an error. It just closes. Using PrusaSlicer 2.6.0 works fine with the same settings.
My Specs are:

AMD Ryzen 9 3900X 12-Core Processor 3.80 GHz
32 GB RAM
AMD Radeon RX 6800

### Project file & How to reproduce

To reproduce the error, try to slice the Moon_Globe_mk2.stl file from this model: https://www.printables.com/model/23859-designer-lithophane-moon-lamp.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1-win64

### Operating system

Windows 10 Version 10.0.19045 Build 19045

### Printer model

Prusa MK4

JohnOCFII, 2023-09-16:
I was able to slice this successfully on an Apple MacBook Pro M1/arm.  It took a while at the 90% point.  I used the default IS Structural settings.

Jan-Soustruznik, 2023-09-22:
Hello, @m00nSec ,

could you share also your project file(3mf), please? We cannot reproduce on your side. 

Thank you, have a nice day

Jan-Soustruznik, 2023-10-24:
Hello, @m00nSec , could you share again, please? Link to download expired. Thank you

manmarko, 2023-11-01:
I am having the same problem. Happens always when bigger objects are sliced. It is also happening with the same file: Moon_Globe_mk2.stl file from this model: https://www.printables.com/model/23859-designer-lithophane-moon-lamp.

Here are some details:
3mf file: [moon-globe-mk2.zip](https://github.com/prusa3d/PrusaSlicer/files/13231542/moon-globe-mk2.zip)
Windows crash reports: [AppCrash_prusa-slicer.exe_d66791c7f745c2233641e6dba08d417aacb1bb_829759b3_444e800a-ea0e-41f6-b37a-1dcec256f02b.zip](https://github.com/prusa3d/PrusaSlicer/files/13231574/AppCrash_prusa-slicer.exe_d66791c7f745c2233641e6dba08d417aacb1bb_829759b3_444e800a-ea0e-41f6-b37a-1dcec256f02b.zip)
Windows event log: [prusa_slicer_error.zip](https://github.com/prusa3d/PrusaSlicer/files/13231583/prusa_slicer_error.zip)
prusa-slicer system info: 
[prusa-slicer-system-info.txt](https://github.com/prusa3d/PrusaSlicer/files/13231593/prusa-slicer-system-info.txt)

```
Processor	13th Gen Intel(R) Core(TM) i7-13700K   3.40 GHz
Installed RAM	32.0 GB (31.7 GB usable)
System type	64-bit operating system, x64-based processor
```

On another PC (much much older and slower) this stl slicing also crashes, but others that are smaller do not crash. The smaller ones always crash on the above PC. 
prusa slicer system info (this one does not crash): [prusa-slicer-system-info-does-not-crash.txt](https://github.com/prusa3d/PrusaSlicer/files/13231619/prusa-slicer-system-info-does-not-crash.txt)

```
Device name	BONE
Processor	Intel(R) Core(TM) i5-4570 CPU @ 3.20GHz   3.20 GHz
Installed RAM	16.0 GB
Device ID	3CA93B13-01DF-4E05-A010-B5C81DD9FF03
Product ID	00330-50000-00000-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display
```

Here are links to the smaller ones that are sliced OK on the second PC:
https://cults3d.com/en/3d-model/art/flexi-print-in-place-skeleton-shark
https://cults3d.com/en/3d-model/art/fidget-spinner-spyder-halloween-spider-articulating-wiggle-pet/comments

closed 2026-09-29 by prusaslicer-bot[bot]