prusa3d/PrusaSlicer #11318: Printer Queue Feature

bertwest, 2023-09-17:
After finding out the possibility to send print jobs directly to a network connected printer I had the following idea:

Wouldn't it be great to have a "printer queue" analogous to what is known from the paper printer queue? 

Imagine:

You have several prints you want to do. You sent the frist, you add the next to the queue, you work on the third and find, that it should print next, so you move the third job to the second place.

The printer gives a notification as soon as a print is finished. Mayby acoustic. Maybe via a notification on the mobile app. 
You go to the printer, remove the finished print, and the printer just asks "continue?" and the next print starts.

I file this issue here being aware that this not only affects prusaslicer but also the printer firmware, but from the user point of view the workflow would start in prusaslicer with the ability to add print jobs to a queue and ultimately having the possibility to manage that queue.

(I did search if such a feature request already exists, but couldn't find anything. Sorry if I missed something.)

sarusani, 2023-09-17:
You can already do this with Prusa Link & Prusa Connect. You'll just need a Raspberry [Pi Zero W / 2](https://help.prusa3d.com/guide/prusalink-and-prusa-connect-setup-mk3-s-_221744) or an [Raspberry Pi 3 / 4](http://help.prusa3d.com/guide/prusalink-prusa-connect-with-rpi-3-4-usb-mk2-5-s-mk3-s-_469341).

bertwest, 2023-09-17:
Thanks! ... Really? OK, in this case I will have to look deeper in Prusa Link and Connect. I did try this, but this feature did not pop out of the UI :-) 
Thanks again.

sarusani, 2023-09-17:
It's a Prusa Connect feature. (But you need Prusa Link to use Connect)
Once you're set up. Just login to Prusa Connect and you'll see "Print queue" in the main menu.

bertwest, 2023-09-17:
Thanks. I found it. 
I now know why it did not pop out to me.
I looked at prusa connect and it really has great features, but I don't like that workflow.
1. Download from printables and store it locally
2. open it in slice in prusaslicer
3. upload to connect
4. download to the printer

Similar when I have my own design.

Why sending all this data around the globe if all relevant devices are 2 meters from each other?

I have a local network that works pretty fast when working with ethernet and a wifi that gives me errors when using prusa connect or prusa link, because it does not like how slow it is in that corners where the printer is located. 

So, I would prefer a local setup. Except if prusa slicer was integrated into prusa connect and printables directly. In this case this cloud solution would allow a integrated end2end workflow. 

Thanks again, my initial question was solved anyway.

closed 2023-09-17 by bertwest

sarusani, 2023-09-17:
Just a small addon... :)
Have you looked at octoprint combined with octoapp (Android/ iOS) . You can send the prints to octoprint out of the slicer, get notifications on your phone when a print is finished, then go to your printer and start the next. (You'll have to press start on your phone instead of the printer )

It's pretty close to what you want and runs locally.