prusa3d/PrusaSlicer #11320: XL Multi Tool - 30 seconds delay/pause to reach target temp on every tool change when using organic supports

salmander, 2023-09-18:
### Description of the bug

When using Prusa Slicer > XL > 5T profile and setting the support extruder to a non-zero

When using Prusa Slicer, Prusa XL multi tool profile and have organic supports enabled and using a specific tool for organic supports, the printer pauses for roughly 30 seconds, every time it tries to change the tool head to the organic support tool head. This delay is caused by the printer trying to reach the nozzle temperature (210) however, this delay does not happen when its going from the organic support tool head to the other tool head. 

I have also tried to print objects with multiple tool heads and this problem does not occur (there is no delay when changing tool heads). See my sample project attached. This problem is increasing the print time by almost a third.

### Project file & How to reproduce

[Shabbir Khan-tree-text-2.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/12644035/Shabbir.Khan-tree-text-2.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer v2.6.1

### Operating system

MacOs 13.5.2

### Printer model

Prusa XL 5T - firmware v4.7.2

salmander, 2023-10-12:
Any updates on this issue?

FidelCapo, 2023-10-23:
SPE-1981

spradlinb, 2023-11-04:
I'm seeing the same with my 2-tool XL. First tool runs without issue, second tool cools when docked, then pauses when moved to the wipe tower as it waits to reach temperature again.

salmander, 2023-12-02:
This issue still exists in the latest XL 5T firmware (v5.1.0) and Prusa Slicer v2.7.0. It is not limited to organic supports or any supports. It happens randomly with any tool heads GCode attached of a test print.
[prusaxl-5-head-toochanging-banding-test_0.4n_0.2mm_PLA,PLA,PLA,PLA,PLA_XLIS_2h43m.bgcode.zip](https://github.com/prusa3d/PrusaSlicer/files/13537490/prusaxl-5-head-toochanging-banding-test_0.4n_0.2mm_PLA.PLA.PLA.PLA.PLA_XLIS_2h43m.bgcode.zip)

salmander, 2023-12-02:
I am not sure anymore if this is a printer firmware issue or a slicer issue. My understanding is that slicer controls when a tool is parked and unparked, so it can look ahead and get the next toolhead to get ready so that when it is its turn, it is ready to be picked up and the printer doesn't have to wait. At least, that is how most prints work but for some reason not all prints exhibit this behaviour. I really hope someone technical from Prusa looks at this. This issue makes the print times completely wild. See screenshot below: 
[image]

u89djt, 2024-05-09:
@FidelCapo Hi there, have you found any workarounds that we can try for this? I'm seeing a few seconds pause for extruder 1 to heat while printing PLA in tool 1 and PETG in tool 2 with this test object I put together to look at interactions between the two filaments.  That's with ooze protection off. XL type B, firmware 6.0 and prusaslicer 2.7.4.
[plapetgmix.zip](https://github.com/prusa3d/PrusaSlicer/files/15266489/plapetgmix.zip)
Best wishes
Dave

closed 2026-09-29 by prusaslicer-bot[bot]