prusa3d/PrusaSlicer #11321: Printer Cooling Multiplier

Kaelum, 2023-09-18:
Different printers have different cooling abilities, and the cooling settings are primarily set in the Filament Settings.  It would be nice to have the ability to specify a cooling modifier in the Printer Settings that would be multiplied by the actual cooling value to allow for normalization across printers.

**Describe the solution you'd like**
Add a setting to the Printer Settings for "Cooling multiplier".

**Describe how it would work**
A value of 1.00 would mean no change.  A value of 0.60 or 60% would mean a reduction of 40% in the calculated cooling setting, or 60% of the calculated value.  A value of 1.25 or "125%" would mean an increase of 25% in the calculated value.

**Describe alternatives you've considered**
Creating duplicate Filament Settings for each printer would be impossible to manage, and is currently the only possible way to manage this.  My Rat Rig cooling fan is at least 50% stronger than any of my other printers.  My Snapmaker is capable of less than 60% of the cooling that my K1 Max is capable of, and my Ender 5 Plus is somewhere in the middle.  As you can see, this is impossible to manage.

rtyr, 2023-09-18:
The correct way is to use separate filament profiles an set compatible printer condition so each printer uses the correct profile. It is not only about cooling. Different printers often needs different temperatures, different retraction overrides, different custom gcodes (linear advance).

Kaelum, 2023-09-18:
That is NOT the correct way, it is currently the ONLY way, as I stated in my post.  The ONLY difference between all of my machines that matters, is the cooling, and there is currently NO POSSIBLE WAY to correct for that.  Creating multiple Print Settings is a waste of time, and impossible to manage, which AGAIN I stated in my post.  That is NOT a viable option.

Kaelum, 2023-12-23:
There is actually a better overall solution, which would address this issue, and several other shortcomings as well.  #11776 is that solution, so I'm going to close this issue, as that solution is much better.

closed 2023-12-23 by Kaelum