prusa3d/PrusaSlicer #11322: Insert Pause option not present at first layer

GaryAitken, 2023-09-18:
### Description of the bug

After slicing, clicking on the first layer does not cause the '+' icon to appear, making it cumbersome to add a pause after printing the nozzle-clearing line.

### Project file & How to reproduce

1. Slice any model.
2. Click on the orange arrow at the very bottom of the layer slider; no '+' icon appears

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

linux 22.04

### Printer model

Prusa I3 MK3S

rtyr, 2023-09-18:
That is intended behavior. There is usually no reason to add color change to the beginning of the first layer. It was initially possible in some alpha/beta versions and it often caused people to misclick on it by accident while they were only trying to move with the slider.

closed 2023-09-18 by rtyr

GaryAitken, 2023-09-18:
There are other reasons to add a pause at the start besides changing color.  In my case, I need to place material, a thin screen, on the bed; the screen is held down by magnets.  The magnets are thin, but they would interfere with the bed-leveling / homing, so have to be placed after the startup sequence is complete.

A misclick can be handled easily by adding a verification step, "Do you really want to ...?", which can be displayed optionally via a configuration setting.  Omitting the capability seems like a radical solution to a misclick.

rtyr, 2023-09-18:
Color change / pause at the start of the print does not make much sense for majority of users.

You can create separate printer profile with modified start g-code for such purpose.