prusa3d/PrusaSlicer #11324: Bridging not working correctly

barnes-dev, 2023-09-18:
### Description of the bug

Trying to bridge that used to work but prusaslicr now prints lines in mid air then connects them to the edge.  It should just draw one continuous line across the bridge or at least draw the end points then connect them later when they cool.  This is a problem for both straight lines and octolinear.
[image]
[image]

### Project file & How to reproduce

using pi3b-hextop.3mf inside raspberry-pi-3b-case-for-noctua-40x10-model_files.zip model from prusaprinters.
[pi3b-hextop-stl.zip](https://github.com/prusa3d/PrusaSlicer/files/12649328/pi3b-hextop-stl.zip)
[pi3b-hextop.zip](https://github.com/prusa3d/PrusaSlicer/files/12649329/pi3b-hextop.zip)

https://www.printables.com/model/18759-raspberry-pi-3b-case-for-noctua-40x10

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Windows 11

### Printer model

MK3S+ with MMU2S

Patronics, 2026-04-26:
I'm encountering the same issue in PrusaSlicer 2.9.4 Slicing for Core One. Thankfully I spotted it in the slicer before the printer got to that step. This design could've been bridge properly (though overhangs may have struggled), but instead it decided to print a floating bridge supported by nothing, and only then attach it to the main structure of the print. 

[image]
Here's a file that reproduces the issue, a particularly egregious example.

[PVC Octagon Lock-FloatingBridge.3mf.zip](https://github.com/user-attachments/files/27093843/PVC.Octagon.Lock-FloatingBridge.3mf.zip)

barnes-dev, 2026-09-18:
> Ditto.

This error has been open for ever.  Has anyone researched what is
happening.  I have a large milticolor print that I have been wanting to
print but it was failing 1/2 way through.  I was waiting for my upgrade to
the booted nozzles but now i am concerned that will not fox the issue
either. It seems that many people have found the same issue with different
models.  There must be a fix somewhere.