prusa3d/PrusaSlicer #11330: x axis slam

ziggyz13, 2023-09-19:
### Description of the bug

first ever time using this slicer and the print starts off ok the origin goes to 0,0 then the x axis slams into the left and keeps trying to go then once it prints its never in the same position. 

### Project file & How to reproduce

[Overhang_Test.zip](https://github.com/prusa3d/PrusaSlicer/files/12655814/Overhang_Test.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

windows 11

### Printer model

anycubic kobra 2

rtyr, 2023-09-19:
It is most likely just misconfigured printer profile. I am afraid we may be not able to help you with configuring your printer. Also, you did not provide the project file.

rtyr, 2023-11-23:
No response. Closing.

closed 2023-11-23 by rtyr