prusa3d/PrusaSlicer #11332: Cannot reload from disk after performing cut

OoDeLally, 2023-09-19:
### Description of the bug

When cutting a piece into two `parts` or `objects`, the ability to `Reload from disk` is greyed out.
Reading this issue https://github.com/prusa3d/PrusaSlicer/issues/2595, I can only assume it used to be possible.
Is that a feature PrusaSlicer no longer support?

This would be very useful to me. Currently I have to redo the cut every single time I update my model.

### Project file & How to reproduce

1. Add a STL.
2. Perform a cut.
3. Right-click on a part > `Reload from disk` is greyed out.

[image]

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.1+linux-x64-GTK3

### Operating system

Ubuntu

### Printer model

Prusa MKS3+

YuSanka, 2023-09-21:
Thanks for your report, but this is expected behavior.
We don't allow to reload cut objects.

closed 2023-09-21 by YuSanka

yahbluez, 2023-10-02:
I second that reload from disk is broken.

Regis-Frollo, 2024-06-18:
> Thanks for your report, but this is expected behavior.
> We don't allow to reload cut objects.

This is foolish behaviour. It is a shame that you dare to write that.

necarlson97, 2026-02-23:
Noting that (as of 2.9.4) - the 'reload from disk' option appears, and a dialogue pops up with the reloading bar - but the model is not actually updated.

[image]