prusa3d/PrusaSlicer #11334: Python Post Processing - Overwrite output file in native path

NGC869, 2023-09-19:
### Description of the bug

Hi,
I'm not sure if the problems encountered results from Prusa Slicer bugs/features, Python capabilities or from my ignorance, anyway...

I'm able to code a fully functional script that read the temporary gCode file ('gcode.pp') and copy it to a new file while modifying it, but **I'm not able to output the desired gCode changes to the Prusa Slicer "native" output file**.

Basically I'm asking why a sample script like the one following won't work. A script that ideally process the temporary gCode file and let the slicer to save it to the native output path/file:

```
import re
import sys
import os.path
from os import path, getenv

sourceFile=sys.argv[1]

input_gcode = open(sourceFile, "W+"):
input_gcode.write("ciao")
input_gcode.close()
```

Yes, I'm also able to obtain the PrusaSlicer native output file/path, using the following code, anyway I'm not able to achieve the desired result.
`ps_output_name = str(getenv('SLIC3R_PP_OUTPUT_NAME'))`

So, which is the correct and best way to manage the output of our post-processing script, also considering the changes introduced by the version 2.4?
Are you able to post a simple example script in order to let us know how to deal with this kind of work?

Thank you so much!

### Project file & How to reproduce

Trying the exposed out sample python script.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Window 10

### Printer model

RepRap Firmware

foreachthing, 2023-09-19:
Yeah, Prusa "screwed-up" the logic how post processing should work, because they want to write to a temp folder first, then save it to the SD-Card. Instead of processing on the SD-Card ... which is actually good, but inconvenient at the same time.

> I'm not able to achieve the desired result.

What is you're trying to do?

You could have a look at this: https://github.com/foreachthing/Slic3rPostProcessing/blob/master/SPP-Python/Slic3rPostProcessor.py
It's not a "small simple" script but you should figure it out.

lukasmatena, 2023-09-20:
@NGC869 

Not that I am some Python expert, but I "w+" and "W+" is not the same and the colon after that command is a syntax error.

I have no clue whatever @foreachthing is talking about.

lukasmatena, 2023-09-20:
@NGC869 I can see that you silently edited the original post and fixed both those issues. As far as I can tell, post-processing the file now works correctly. Can we close the issue?

foreachthing, 2023-09-20:
Does this help?
https://blog.gitnux.com/code/python-overwrite-file/#:~:text=To%20overwrite%20a%20file%20in%20Python%2C%20you%20can%20use%20the,which%20effectively%20overwrites%20the%20file.

NGC869, 2023-09-20:
Hi all and thank you for your reply!

@foreachthing I need to modify the gCode by some simple conditional expression (eg. reading it line by line), then save the modified gCode overwriting the Slicer native output file in the native output path. Ideally an easy task, but it seems more hard to code the saving part respect coding the conditional expression :)
In my opinion a Slicer should just care to produce a good gCode, not care where the gCode will be saved (the SD card problem's could be faced by some post-processing script.)
Anyway I'll look at your code ;-)

@lukasmatena I've corrected the code, but not silently because I am answering :-) Let me show some code.

lukasmatena, 2023-09-20:
> not silently because I am answering

Ok, I thought that you ended with the edit.

This is an example script which will modify the file in place and convert it to lowercase.
```
import re
import sys
import os.path
from os import path, getenv

sourceFile=sys.argv[1]

input_gcode = open(sourceFile, "r")
lines = input_gcode.readlines()
input_gcode.close()

lines = [line.lower() for line in lines]

output_gcode = open(sourceFile, 'w')
output_gcode.writelines(lines)
output_gcode.close()
```

Closing.

closed 2023-09-20 by lukasmatena

NGC869, 2023-09-20:
The working (in PS 2.6) following example code

lukasmatena, 2023-09-20:
@NGC869 To be honest, I don't know what else than "Post processing scripts shall modify G-code file in place" you need. But I am glad that the problem is solved.