prusa3d/PrusaSlicer #11335: anomaly in slicer/g-code

brizp, 2023-09-19:
### Description of the bug

There's an unexpected and non-functional deviation in the fill-pattern/top layers on one of the four pilon's in the middle

### Project file & How to reproduce

Just load the project and slice.

There's an unexpected and non-functional deviation in the fill-pattern/top layers on one of the four pylon's in the middle.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0+MacOS-x64 / PrusaSlicer-2.6.0+MacOS-x64-202306191415

### Operating system

osx10.12.6

### Printer model

prusa i3 MK3S

neophyl, 2023-09-19:
No project file attached. You need to zip files to attach them.

brizp, 2023-09-19:
Strange, I had included them ~, this time using a different browser.
Hope it works now.
[case.zip](https://github.com/prusa3d/PrusaSlicer/files/12665099/case.zip)

brizp, 2023-09-19:
I created a step file instead of a stl and the step slices correct.

neophyl, 2023-09-19:
Some of the vertices around the hole are fractionally off from the others by the time the layer is worked out so its splitting what should be a flat face between 2 layers.  

I used the place to bed function on the big sloping section to in effect flip the part over and then placed the bottom back to the bed using place to bed again to force the bottom to be flat to the bed, and sliced and that made the top surface slice properly.
[image]

Also changing the layer height so all the verts fall into one layer or another also worked before I flipped it.

Theres some really heavy geometry all converging on those holes for some reason, far more than is needed which I think is contributing to the problem.

Its basically mentioned in the FAQ https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ)

brizp, 2023-09-19:
Ah, indeed, using place to bed like you said solves the issue.
Thank you!

closed 2023-09-19 by brizp