prusa3d/PrusaSlicer #11336: Negative modifier - problem with supports

TheElesty, 2023-09-19:
### Description of the bug

When you remove part of model with negative modifier, it generates supports under that cuted part. Second image crops bottom 50~ layers for better visibility.

This model is not my , try of remixing it to has rubber grips and easy way was to cut that part and then use it as neg. mod on original model. To cut i used meshmixer, but it should not be relevant for this problem.
Miyoo Mini Plus Grip by JohnyBoy2187 on Thingiverse: https://www.thingiverse.com/thing:6061286

### Project file & How to reproduce

[image]
[image]

[Grip_remix.zip](https://github.com/prusa3d/PrusaSlicer/files/12666158/Grip_remix.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

Win10  64bit

### Printer model

MK3s+

TheElesty, 2023-09-19:
I resliced it with organic supports and they behave weardly too

FidelCapo, 2023-09-21:
The problem is probably due to the fact that the Negative Volume intersects the model very closely. If the Negative Volume exceeds the Supports by even a little bit they are generated correctly. The same problem occurs when trying to export STL in a boolean operation..

closed 2026-09-29 by prusaslicer-bot[bot]