prusa3d/PrusaSlicer #11337: First Layer G-code errors.

noobycoder9, 2023-09-20:
### Description of the bug

The first layer is rendered as two horizontal shells layered on top of each other. Same issue with any STL.
[image]

https://github.com/prusa3d/PrusaSlicer/assets/120605300/eb968518-3e5b-42d5-bf08-18ca75670c6d

### Project file & How to reproduce

[zyncarrierpouchproject.zip](https://github.com/prusa3d/PrusaSlicer/files/12667995/zyncarrierpouchproject.zip)
Working with any STL, not necessarily this project. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

windows10

### Printer model

Ender3

neophyl, 2023-09-20:
The problem setting is the -0.25 value you have under Printer Settings>General>Z Offset.  put that back to 0 and it slices and previews fine.
It might be only a slice preview issue as the gcode produced looks like its using z=0 for the first layer and z=.25 on the second layer.  Have you actually tried printing ?

noobycoder9, 2023-09-20:
> The problem setting is the -0.25 value you have under Printer Settings>General>Z Offset. put that back to 0 and it slices and previews fine.
> It might be only a slice preview issue as the gcode [...]

I haven't tried printing it yet but I can soon. I've had the z offset like that for quite some time and it's only now caused the issue. Might be that in combination with another setting I recently changed. Will try to print in the next few days.

kubispe1, 2023-09-21:
SPE-1902

lukasmatena, 2023-10-21:
Fixed in [PrusaSlicer 2.7.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-alpha1). Closing.

closed 2023-10-21 by lukasmatena