prusa3d/PrusaSlicer #11342: Nefunkční kontrola aktualizace

rjkrause, 2023-09-21:
### Description of the bug

Při pokusu o kontrolu aktualizace programu je zobrazeno hlášení.
Do Slicer-u neni možné nainstalovat Mini Input Shaper

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.1+win64-202309060711

### Operating system

 Windows 10 (sestavení 19045), 64bitová edice

### Printer model

Prusa Mini

Jan-Soustruznik, 2023-09-22:
Dobrý den, @rjkrause 

Objevuje se chyba pokaždé, když chcete aktualizovat profil? Pokoušel jsem se vyvolat stejnou chybu a nepodařilo se mi ... je možné, že se mohlo někde něco cestou pokazit, server mohl být zahlcen atd ... 

Popřípadě bych ještě doporučil zálohovat profily a přeinstalovat slicer. Pokud by chyba přetrvávala. Dejte nám vědět, prošli bychom problém více detailněji. 

Děkuji a přeji hezký den

rjkrause, 2023-09-25:
Dobrý den,
ano, chyba je trvalá. Přeinstalace Slicer-u nepomohla. Napadá mne souvislost s tím, že SW je instalován na firemním stroji, kde jsem nepotvrdil hlášku o povolení akce na firewalu. Bohužel nedokážu navodit stav, abych tuto hlášku zobrazil znovu. 
Pokud zkopíruji link co se  nabízí ve výše uvedeném hlášení a zkusím tento načíst do prohlížeče skončí to chybou také. Možná je problém s ověřením...

Jan-Soustruznik, 2023-09-25:
Dobrý den, 

povolení aplikací v rámci firewallu by mělo být(WIN10) v _Nastavení > Akualizace a zabezpečení > Zabezpečení Windows > Firewall a ochrana sítě  > Povolit aplikaci v bráně firewall // Obnovit brány firewal na výchozí nastavení_

Každopádně pokud používáte firemní stroj, určitě bych před jakoukoliv změnou nastavení firewallu kontaktoval vaše IT oddělení.

Přeji hezký den

rjkrause, 2023-09-27:
Dobrý den,
Problém s nastavením jsem vyřešil, nové nainstalovaná aplikace sliceru na
novým stroji sice detekuje že je vše aktuální y nejméně v konfiguraci stále
nevidím možnost přidat profil tiskárny MINi s rozšířením Input shaper - což
mělo být výsledkem mého trápení, kdy jsem nainstaloval firmware pro Mini s
Input Shaper, ale vlastně se mi nedaří slicovat model pro toto rozšíření.
Se starým slícováním tiskárna prochází resetem do modré obrazovky ( to už
se všude řeší jako chyba,ale zatím pro to není dostatečná odezva s
vyřešením ze strany podpory od Vás,co by výrobce ) . Po tomto update mám
vlastně s tiskárny cihlu na stole...

Co dál ,

Děkuji

Jan-Soustruznik, 2023-09-27:
@rjkrause , 

Dobrý den, 

Nové profily mini s IS teď musíte přidat, tzn _Konfigurace > Průvodce nastavením.. > (zde se může zobrazit výzva k nainstalování chybějících konfigurací - pokud se objeví tak doporučuji instalovat) > (Poté co se načte průvodce) Prusa FFF > MINI Rodina a vybrat profil označený jako ALPHA_

closed 2026-09-29 by prusaslicer-bot[bot]