prusa3d/PrusaSlicer #11345: "Arrange" not working properly

mwitarsa, 2023-09-21:
### Description of the bug

Arrange function that organize multiples objects on the plater is not working when "complete individual objects" is checked (from "output options" on "print settings"

### Project file & How to reproduce

[Arrange not working.zip](https://github.com/prusa3d/PrusaSlicer/files/12693923/Arrange.not.working.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.1+win64

### Operating system

Windows 11 Pro

### Printer model

Creality CR-200B

Jan-Soustruznik, 2023-09-22:
Hello @mwitarsa , 

could you provide more details(steps, video)? What do you means not working properly? What happend when you did arrange?  

I opened your 3mf file and arranged the model ... I dont see any problem? All models were arranged with requested offset. 

What setting of arrange do you use? fast, Balanced,Accurate?

Do you have this problem? https://github.com/prusa3d/PrusaSlicer/issues/11340

mwitarsa, 2023-09-28:
Did you try with "complete individual objects" checked (from "output options" on "print settings")?

mwitarsa, 2023-09-29:
> Hello @mwitarsa ,
>
> could you provide more details(steps, video)? What do you means not working properly? What happend when you did arrange?
>
> I opened your 3mf file and arranged the model ... I dont [...]

Yes, I'm having exactly this problem #6474 [https://github.com/prusa3d/PrusaSlicer/issues/6474](url) and #11340 [https://github.com/prusa3d/PrusaSlicer/issues/11340](url)

lukasmatena, 2023-10-04:
Thanks. Closing as a duplicate of #11340 then.

closed 2023-10-04 by lukasmatena