prusa3d/PrusaSlicer #11348: i found new cool thing but i need to color change 3x in a layer can you add it to the bottom lever like the right

Ramonline, 2023-09-22:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

multi color change in 1 layer  is very difficult and i should do this ez by editing gcode left or add it by lever at bottom like i do on the right lever

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

iam making stuff marble contest of polymaker and i wanna color the infill diffrend 4x in a layer
beside painting what does not work out for me

**Describe how it would work**
Be as descriptive as your skill level allows you.

 it should be great if i could use the step lever the same as the layer lever on right same options would fix it just + color switch or edit gcode on point you are if you move the steps levers

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

asking big ppl like papa jo follows and they point at paint but its not 100% doing what i like 

mean i go where you switch colors

now infill starts spiral say 1/3 of this change color 2/3 of 1 layer      3x a layerhow can i do this ez like the way i write i think

you all gonna see it  at polymaker stand next week

**Additional context**
Add any other context or paint drawing about the feature request here.

rtyr, 2023-09-22:
There is already FR for this, I am closing this as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5195

closed 2023-09-22 by rtyr

Ramonline, 2023-09-22:
so thats 2 years ago think its forgotten put it on again then pls

Ramonline, 2023-09-22:
i do not agree it is already tried and i do not think its allot of work to add it because its already there but not programmed 100%