prusa3d/PrusaSlicer #11350: simple edit gcode and iam not gonna write allot it should be in the slicer period

Ramonline, 2023-09-22:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

closed 2023-09-22 by rtyr

rtyr, 2023-09-22:
The description is missing and the title does not help. If this issue was not created by accident, you can add the description and we can reopen it (if it is not a dupliate request).

Ramonline, 2023-09-23:
can prusa pu please edit gcode in slicers its been in replicator or repertier already from 2014 why is this not in there

Ramonline, 2023-09-23:
mean i can do more then push a button so expert should have it

rtyr, 2023-09-23:
https://github.com/prusa3d/PrusaSlicer/issues/5951 

https://github.com/prusa3d/PrusaSlicer/issues/5195